noble/noble #553: cloned noble did not work with npm install

OwenBrotherwood, 2017-02-13:
I cloned nobel to continue with understanding bluetooth.
I had previously cloned node-bluetooth-hci-socket, which did not complain about missing pre-req or bluetooth adapter.

However, with npm install followed by node examples\advertisement-discovery.js, I got a 

`    throw new Error('No compatible USB Bluetooth 4.0 device found!');`

npm install ..\node-bluetooth-hci-socket and another attempt gave me

```
peripheral discovered (dff5a5ec431b with address <df:f5:a5:ec:43:1b, random>, connectable true, RSSI -67:
        hello my local name is:
                Puck.js 431b
        can I interest you in any of the following advertised services:
                ["6e400001b5a3f393e0a9e50e24dcca9e"]
```

Hello dear puck, where do we go now: windows 10 with Zandig and noble are working.

OwenBrotherwood, 2017-02-14:
And with espruino/EspruinoWebIDE

sandeepmistry, 2017-02-16:
What adapter are you using? I haven't done a `node-bluetooth-hci-socket` release in a while, so master is not in sync with what's on npm.

OwenBrotherwood, 2017-02-16:
Intel Wireless Bluetooth 7265	0x8087	0x0a2a

OwenBrotherwood, 2017-02-16:
This may be something I need to open as a new issue.
Research in https://github.com/espruino/EspruinoWebIDE/issues/168#issuecomment-280275602 has now resulted in a thrown error for noble when noble is required.
Please note, in the environment, Windows 7 Zadig, the example works inside the node_modules tree for the requiring code:
 
`Error: Systemet kan ikke finde meddelelsesteksten for meddelelsesnummer 0x%1 i meddelelsesfilen for %2.
`

```
C:\source\repos\github\espruino\EspruinoWebIDE\node_modules\noble>node examples\
advertisement-discovery.js
```

OwenBrotherwood, 2017-02-17:
Windows 10
I am not sure if you will hit problems but the following for node 7.5.X was a killer for me as I used so much time finding out why I could not make a simple npm install usb on 7.5: I have reverted back to node 6 LTS and reassess the damages
https://github.com/tessel/node-usb/issues/163

OwenBrotherwood, 2017-02-18:
https://github.com/espruino/EspruinoWebIDE/issues/168#issuecomment-280861201 Windows 10 success with Gordons https://github.com/espruino/winnus in a bluetooth only with no usb scenario.

usb is proving to be a nasty for me and has caused a downgrade of node and nw.js because of https://github.com/tessel/node-usb/issues/163

Gordons IDE and node-usb issue gives the following setup that I will now try to achieve a positive result on and then go back to Windows 7 testing.
node 6.8.1 https://nodejs.org/download/release/v6.8.1/
nw.js https://nwjs.io/blog/v0.18.2/ (based on node 6.8.1)
Github clones from Gordons IDE and from nobel and node-bluetooth-hci-socket

I believe I have mismatch with the current LTS node so I am downgrading to the node 6.8.1 and work on it "now".

I  will edit this comment with results

sandeepmistry, 2017-02-21:
> Intel Wireless Bluetooth 7265	0x8087	0x0a2a

Support for this was added in https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/ae33b4f58bf4209f6545e7ec60f3c5015196d2db (Jan 15) which has not been published to npm yet. The current version published on npm is  0.5.1 which was published Sept 6th, 2016.

This should be resolved with the next npm release of `node-bluetooth-hci-socket`

fedyfausto, 2018-11-28:
i get the same prolem but i can not fix with the Zadig

Intel Wireless Bluetooth 8265 | 0x8087 | 0x0a2b