noble/noble #556: Why connectable state never changes ?

THLN47, 2017-02-21:
Hello,
I've some BLE devices. The advertisement works properly except the "connectable" status.
this parameter is still set to false even if the device becomes connectable.

function onDiscover(peripheral) {
    var adv = peripheral.advertisement;
    var uuids = adv.serviceUuids;
	var connectable = peripheral.connectable;
	console.log(connectable);
}

hcidump :

> HCI Event: LE Meta Event (0x3e) plen 42
    LE Advertising Report
      ADV_NONCONN_IND - Non connectable undirected advertising (3)
      bdaddr 94:54:93:0C:D5:56 (Public)
      Flags: 0x05
      Unknown type 0x07 with 16 bytes data
      Unknown type 0xff with 7 bytes data
      RSSI: -69
...
> HCI Event: LE Meta Event (0x3e) plen 42
    LE Advertising Report
      ADV_NONCONN_IND - Non connectable undirected advertising (3)
      bdaddr 94:54:93:0C:D5:56 (Public)
      Flags: 0x05
      Unknown type 0x07 with 16 bytes data
      Unknown type 0xff with 7 bytes data
      RSSI: -55
> HCI Event: LE Meta Event (0x3e) plen 30
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 94:54:93:0C:D5:56 (Public)
      Unknown type 0x19 with 2 bytes data
      Flags: 0x06
      Complete local name: 'toto ....'
      RSSI: -54
> HCI Event: LE Meta Event (0x3e) plen 30
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 94:54:93:0C:D5:56 (Public)
      Unknown type 0x07 with 16 bytes data
      RSSI: -56

mattgol, 2017-02-22:
The status "connectable" is only set when the device is is discovered for the first time and not changed afterwords. In order to change there are two changes necessary:
1. The environment variable NOBLE_REPORT_ALL_HCI_EVENTS has to be defined. Otherwise the "nonconnectable" advertisements will not be reported to the upper levels of noble. 
2. The status has to be assigned to the peripheral object with every advertisement. To do this add the this line to line 126 of noble.js: `peripheral.connectable = connectable;`
No connectable will always reflect the current status of the peripheral. 
Good luck!

THLN47, 2017-02-23:
Thanks for your reply "mattgol",
I've added the code you've shown, and now the connectable status detection is working.
When my device enters in connected mode, my node.js program can now establish the connection.

But why this modification is not in the noble.js program yet ? 
Is there some problem with it ?

mattgol, 2017-02-23:
I don’t want to speculate on the full extent of the design decision behind this choice but there is some logic behind it. Maybe I should elaborate a little bit how advertisements are handled in noble.

For BLE advertisement there are actually more choices than just connectable or nonconnectable: There is also direct/indirect advertising and scannable/not scannable. Noble shields the user from most of this, only percolates some data to the top, and filters out some other advertisements:
- If the device is scannable the dongle automatically requests a scan response with further information about the device (active scanning). 
- For scannable device noble waits for the scan response and does not report the advertisement packet itself. (According to the Bluetooth standard scannable means always also connectable.)
- If there is no scan response for some reason (lost packet etc.) we get the second advertisement packets that comes in and any further packets until there is an actual scan response.
- This means if noble already received a scan response at any point in time from the device the standard behavior is only to emit response events. The idea is that scan responses contain more information and we get more consistent data about the device by only looking at the scan responses. 
- This also means if the device later decides to send out nonscannable/nonconnectable advertisements noble ignores them.
- In order to get every advertisement packet the flag NOBLE_REPORT_ALL_HCI_EVENTS is needed. Then we get any advertisement event. 
- When the first advertisement/scan response is bubbled up noble also constructs the peripheral object and assigns the connectable flag. Later this flag is not touched again unless we add the extra line from the comment above into noble.js.

All in all this means that noble doesn't handle your case very well because of the preference of scan responses over advertisements. Even with line 126 as proposed we can only see the change from non- to connectable but not the other way around if there's a scan response at any time in the lifetime of running noble instance. 

Almost all the devices I’ve encountered so far don’t switch back and forth between connectable and nonconnectable – only my toothbrush changes to nonconnectable when I start brushing which doesn't clash with current noble behavior as the first packet will be connectable.
It's up to Sandeep to decide how to handle this case.

THLN47, 2017-02-24:
Thank you for your explanations.

I'm programming a BLE device firmware with Nordicsemi MCU (nRF52832).
My BLE device had two modes : In the first one, it advertises (ADV_NONCONN_IND) data in the manufacturer data section. And when I push the connect button on my device, it flips into the second mode. At this time, it advertises (ADV_IND) to indicate scannable feature to the BLE client. 
This one can also send ADV_SCAN_REQ requests, and my BLE device responds by a ADV_SCAN_RSP frame.

My node.js program processes and shows manufacturer data in non connectable mode until my BLE device flips into the connnected mode. The dectection of the connectable state allows to establish the connection. 

I saw that the connectable state doesn't return to false after becomming true.  But in my case, I don't care about this behavior.

In fact, at the end of the connection, the "noble.startScanning()" function should be called to receive news advertisement frames. I guess that the connectable state is reseting to false.

mattgol, 2017-02-24:
I played around a little bit with my toothbrush and found another setting that is necessary: When startScanning is called allowDuplicate should be set to true. Otherwise the status is not updated correctly. Only if scanning is started with noble.startScanning([],true) I always get the correct status update. Otherwise the behavior is erratic and depends on which packets are received first. I updated my first comment to include this.

I now changed my noble copy to always report ADV_DIRECT_IND and ADV_NONCONN_IND packets. This way I don't need the NOBLE_REPORT_ALL_HCI_EVENTS any more. I changed line 235-236 in gap.js to this:
`
if (type === 0x01 || type === 0x03 || type === 0x04 || (discoveryCount > 1 && !hasScanResponse) || process.env.NOBLE_REPORT_ALL_HCI_EVENTS) {
    this.emit('discover', status, address, addressType, connectable, advertisement, rssi);
  }
`

THLN47, 2017-02-28:
Hi,

Well Done! I've tested with the latest modification (On Windows this time), and the connectable state is well updated in the both way.

sandeepmistry, 2017-03-13:
@THLN47 @mattgol please open a pull request with the desired changes to continue the discussion ... thanks.

mattgol, 2017-03-13:
@sandeepmistry 
Sure! I'll send a pull request with these changes - give me some time to isolate everything into a separate branch.

ccamrobertson, 2017-08-04:
@mattgol testing and working here, thanks. Did you have the chance to get those changes into another branch for a PR?

lilliesAndRoses, 2018-04-20:
Is there any plan to get these changes into main branch - maybe based on a flag?