noble/noble #560: can't receive any notifications

Shinji0, 2017-03-01:
Hello.

I am using my laptop with win10 and external bluetooth adapter as central. I have nrf51 development kit as peripheral and i am following this tutorial to get more familiar with BLE. 
[https://devzone.nordicsemi.com/tutorials/17/](url)

Everything went fine (discovering services, characteristics, reading characteristics) until i tried to get some notifications using this code:
```
noble.on('discover', function(peripheral) {
  peripheral.connect(function(err) {
    console.log('connected to peripheral: ' + peripheral.uuid);
    peripheral.discoverServices(['0000f00d1212efde1523785fef13d123'], function(err, services) {
      var TempService = services[0];
      console.log('discovered Temperature service');
      TempService.discoverCharacteristics(['0000beef1212efde1523785fef13d123'], function(err, characteristic) {
        var temp = characteristic[0];
        console.log('discovered Temperature characteristic');
        temp.on('read', function(data, isNotification) {
          console.log('Temperature Value: ', data.readUInt8(0));
        })
        temp.notify(true, function(error) {
          console.log('temperature notification on');
        })
      })
    })
  })
})
```

In console i get the following:
```
scanning...
connected to peripheral: c5714dd8aaf6
discovered Temperature service
discovered Temperature characteristic
temperature notification on
```

... and nothing more despite the changes of the temperature value on the nrf51.

Any ideas what is the problem here?

ookjosh, 2017-03-01:
I had a similar issue. Try using `temp.on('data', ...)` instead of 'read'. 

Per the readme (for `characteristic.subscribe`), "Subscribing to a characteristic, triggers 'data' events when peripheral sends an notification or indication". It appears that this function has been renamed to `characteristic.notify` (like you are using) but the docs haven't been updated.

Hope this helps.

Shinji0, 2017-03-02:
Thanks! 
It worked with the 'data'.

Best regards

sandeepmistry, 2017-04-17:
I think this regression was introduced in https://github.com/sandeepmistry/noble/commit/2d343be12044efba12944d9eb39a9d3f31edde9f

I'll take a look at fixing it now ...

sandeepmistry, 2017-04-17:
I've reverted https://github.com/sandeepmistry/noble/commit/2d343be12044efba12944d9eb39a9d3f31edde9f and pushed a fix in https://github.com/sandeepmistry/noble/commit/f66ba04575c901f7c9aa35598c2b163c16c32277.

You can test out the latest via `npm install sandeepmistry/noble` - I'll also update the wiki to use the non-deprecated API's.

closed 2017-04-17 by sandeepmistry