noble/noble #561: Install failed on Raspberry PI 3

raf6410, 2017-03-07:
Hello, 

I tried to install "node", so i follow instruction to install but some probleme happened during intall process.
could you help me ?

Thanks

RAF6410

pi@raspberrypi:~/node $ sudo npm install noble

> usb@1.2.0 install /home/pi/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v51-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@7.7.1 (node-v51 ABI) (falling back to source compile with node-gyp)
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/7.7.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/usb/.node-gyp"
make: Entering directory '/home/pi/node_modules/usb/build'
make: *** No rule to make target '../.node-gyp/7.7.1/include/node/common.gypi', needed by 'Makefile'. Arrêt.
make: Leaving directory '/home/pi/node_modules/usb/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:194:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.34-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/pi/node_modules/usb/src/binding"
gyp ERR! cwd /home/pi/node_modules/usb
gyp ERR! node -v v7.7.1
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:194:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:899:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.34-v7+
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/pi/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node_modules/usb
node-pre-gyp ERR! node -v v7.7.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/node_modules/usb/src/binding' (1)
/home/pi
└── noble@1.8.0

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@~0.1.4 (node_modules/noble/node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.2.0 (node_modules/usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.2.0 install: `node-pre-gyp install --fallback-to-build`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

mattgol, 2017-03-07:
Hi! 

Installing on a Raspberry worked for me without a problem. Did you install all packages 
 via `sudo apt-get install bluetooth bluez libbluetooth-dev libudev-dev`

raf6410, 2017-03-08:
Hi 

Thanks for you response.
Yes i already install this packages.
I search and i have found that "https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v51-linux-arm.tar.gz" don't work => ERROR 404

So how can do ?

mattgol, 2017-03-08:
This used on work on Raspbian until recently - it still works on mine but I haven't installed a new image for a while. Anyhow, I popped in a new SD card and tried the latest release and could fully replicate the behavior you described.

The missing file isn't the real reason behind all this: node-usb only provides binaries for Windows and Mac. After the failed download the installer automatically defaults to compiling node-usb from the source. And here's where the actual problems occur: `No rule to make target '../.node-gyp/7.7.1/include/node/common.gypi', needed by 'Makefile'. `  Googling this error points in the right direction: npm doesn't give root the right permissions by default. 

The solution depends on what you'd like to do: If you only need a local copy of noble then drop the sudo and do the installation under the normal user. (sudo is needed afterwards to set the permissions to access the hardware).

If you still want to do everything with sudo it's possible with the option --unsafe-perm:
`sudo npm install noble --unsafe-perm`
Anyway, sudo is probably only needed for global installation.

Good luck!

P.S. Compiling node-usb will show a lot of warnings but it still works.

raf6410, 2017-03-08:
thanks, it ' works
the installation is OK

Thank you for your help.

sandeepmistry, 2017-03-13:
@mattgol thanks for helping out!

I'm closing this as resolved.

closed 2017-03-13 by sandeepmistry

thomaslc66, 2019-07-22:
This is still happening in 07.2019 and need to be reopened as noble doesn't work with node 10 on raspberry pi3 and on other devices

zaabalonso, 2019-12-28:
I have the same issue @mattgol this needs to be reopened.

Node: 10.16.3 and RSP3b+