noble/noble #563: Is there a disconnect-event before connecting?

DerpDerpson, 2017-03-09:
Hello,

I write an application, which dynamically connects and disconnects to bluetooth devices.
Therefore I need to know, if the devices can "see" each other.
If I register the disconnect-callback on a not connected device, the callback will never be called.
If I call peripheral.updateRSSI() on a not connected device, the callback will never be called and I won't get an rssi-value or an error.

The only information I get, come from the discovery-event, but the values don't come very often
(I think it helps to call noble.stopScanning() and noble.startScanning in an interval). Is it possible to check the peripheral and see if it is still in the range of my Bluetooth-Adapter?

My NodeJS-Application is written in TypeScript, runs on an RPI3 and looks like this:
`import noble = require("noble");
class NobleTest
{
	private knownDevices: string[] = [];

	constructor()
	{
		noble.on("stateChange", (a_state: string): void => 
		{
			if (a_state === "poweredOn")
			{
				console.log("start scanning");
				noble.startScanning([], true);
			}
		});

		noble.on("discover", (a_peripheral: noble.Peripheral): void =>
		{
			if (a_peripheral.address === "unknown") // for dev on MacOS
				return;

			console.log("Discovery: " + a_peripheral.address + " - rssi: " + a_peripheral.rssi);

			if (this.knownDevices.indexOf(a_peripheral.address) >= 0)
				return;

			this.knownDevices.push(a_peripheral.address);

			a_peripheral.once("disconnect", (a_error): void =>
			{
				console.log("Disconnect: " + a_error);
			});
		});
	}
}
new NobleTest();`

mattgol, 2017-03-10:
Hi!

Your callback will be called - eventually...
The callback for disconnect is only called after an actual disconnect occured. Internally, the noble library feeds the callback into an event handler that waits for the a disconnection event and starts the disconnection procedure. If the device is not connected before  the disconnection event does not occur and the callback is not called. It will get called eventually when the device gets connected and disconnected later.

Regarding the RSSI: During scanning the RSSI is updated with every advertisement package received. So for an unconnected device there is no need to call updateRssi. If you do call the function you end up in the same situation as with disconnect: The callback still waits until it finally receives an rssi value during the next connection.

I don't think this is a bug: Noble does not perform any checks and it's up to the application to check the connection status.

Regarding checking if a device is still in range: Without a connection the advertisements (received by 'discover') are the only communication channel available. You would get better results with keeping a connection open to monitor if devices leave the range. I played around using the advertisement from a key tag for presence detection and at least once per day the signal is lost for almost a minute.

The reason that not every advertisement is received is probably based in the wireless connection itself as some packet simply get lost because of noise, devices sending out packets at the same time, or simply the receiver not tuned to the right channel at the right time. In my experience it's never necessary to switch of scanning in regular intervals - I had my application scan for several days without switching it off. 

Hope this helps...

sandeepmistry, 2017-03-12:
@DerpDerpson 

> Is it possible to check the peripheral and see if it is still in the range of my Bluetooth-Adapter?

Please see the enter-exit example :https://github.com/sandeepmistry/noble/blob/master/examples/enter-exit.js - that should be close to what you are looking for.

closed 2017-03-12 by sandeepmistry

DerpDerpson, 2017-03-13:
Thank you for your replies.

In the example @sandeepmistry stated, the exit period is at 2000 ms.
That would mean, that I'd have to get at least 1 discovery per device in 2s. 
But sometimes, when I have multiple devices connected (and write/readHandles and updateRssi in an interval) I don't get any rssi-values for some devices for a minute or more.
Is it normal to get less discoveries, when you "stress" the module? 
Would it help to stop all write/readHandle und updateRssi calls in an interval (like 1s every 10s) to get discovery-information for one second?
Should I serialize all these calls, or is it possible to call these functions parallel?

mattgol, 2017-03-13:
Now you got me confused. Are you connecting to your device and call updateRssi or are you simply scanning for advertisements whild other devices are connected? In the latter case the bluetooth radio actually has to divide its time between the connection and receiving advertisements. This might explain the less frequent advertisements you see. You could try a slower connection interval (try the branch connection-interval-test) which reduces the load on the radio a bit.

DerpDerpson, 2017-03-15:
I am scanning for advertisements, and at some point I might decide to connect to a device.
When I am connected to a device, I will call updateRSSI in an interval of 1s. When the RSSI gets too low, I will disconnect the device. 
Meanwhile I am still scanning for advertisements and connected or connecting to other devices.

The problem seemed to be, that I had too many devices connected and called read/writeValue too often. 
Now that I have decreased the amount of read/writeValue calls I get more advertisements and can work with the enter-exit example to see, if a device is visible.

sandeepmistry, 2017-03-26:
This is mostly like a limitation of the Bluetooth 4.0 adapter you are using.