noble/noble #565: userDescriptionDescriptor.readValue is not a function

fshirzadi, 2017-03-12:
Getting not a valid function error: userDescriptionDescriptor.readValue(function(error, data)
when running: noble/examples/peripheral-explorer.js

That's with async installed.  

Different issue seems to indicate "error" is not a valid parameter being passed.
https://github.com/securing/gattacker/commit/d335b80a22d334112961659b00cd80d342b6f54d

sandeepmistry, 2017-03-13:
@fshirzadi is this an issue with the websocket binding? If so could you submit a PR for https://github.com/securing/gattacker/commit/d335b80a22d334112961659b00cd80d342b6f54d. Thanks.

fshirzadi, 2017-03-15:
I gave up on that example as it seemed to be a few years old and wasn't sure if it was outdated.

Used the Pizza Central.js example and was able to Subscribe, Read, etc... For others reading, BLE Device firmware may have Characteristic or Descriptor which will throw an error when read. It helps to target a known Service, Characteristic,etc. to perform Read/Write operations and narrow down errors.

sandeepmistry, 2017-03-26:
@fshirzadi this doesn't really help us, please  provide detailed steps to reproduce this condition and re-open the issue.

closed 2017-03-26 by sandeepmistry