noble/noble #566: Cannot detect discover configuration service

prabhakarlad, 2017-03-13:
I have following .js, In which I just want to enable notification on for 2902 service, but when I run this application I don’t get 2902 service at all. But Using the gattool I get this service. Any pointers what could be missing.

var noble = require('noble');
var remoteServiceUuid = '180d';

var charuuid = '2a37';
var HRMChar = null;
var hrmServiceNotify = '2902';
var HRMServiceNotifyChar = null;
var srvuuid = '180d';
var fmt = 1;

/* Scan for the custom service ID used by the remote */
noble.on('stateChange', function(state) {
	if (state == 'poweredOn') {
		noble.startScanning([remoteServiceUuid]);
	} else {
		noble.stopScanning();
	}
});

noble.on('discover', function(peripheral) {
	console.log('Found '+peripheral.advertisement.localName+' at '+peripheral.address);

	peripheral.connect(function(err) {
		if (err) {
			console.log('Could not connect: '+err);
			return;
		}

		console.log ('Connected');

		peripheral.discoverServices([remoteServiceUuid], function (err, services) {
			/* Make sure we only handle discovery once */

			services.forEach(function(service) {
				console.log ('Found service: '+service.uuid);
				service.discoverCharacteristics(null, function(err, characteristics) {
					characteristics.forEach (function(characteristic) {
						console.log ('Found characteristic: '+characteristic.uuid);
					});
				});
			});
		});
	});

	peripheral.once('disconnect', function() {
		console.log ("Disconnected");
		noble.startScanning([remoteServiceUuid]);
	});
});

Services & Characteristic detected:
Found TICKR A231 at cb:d2:3c:b2:03:1b
Connected
Found service: 1800
Found service: 1801
Found service: 180f
Found service: 1800
Found service: 1801
Found service: 180f
Found service: 180a
Found service: a026ee010a7d4ab397faf1500f9feb8b
Found service: a026ee030a7d4ab397faf1500f9feb8b
Found service: 180d
Found characteristic: 2a00
Found characteristic: 2a01
Found characteristic: 2a04
Found characteristic: 2a29
Found characteristic: 2a27
Found characteristic: 2a26
Found characteristic: 2a05
Found characteristic: 2a19
Found characteristic: a026e00a0a7d4ab397faf1500f9feb8b
Found characteristic: 2a37
Found characteristic: 2a38
Found characteristic: a026e0020a7d4ab397faf1500f9feb8b
Found characteristic: a026e0040a7d4ab397faf1500f9feb8b

mattgol, 2017-03-13:
Hi!

The UUID 0x2902 is reserved for the descriptor "Client Characteristc Configuration" which switches on notification of the characteristic it belongs to. To discover it you would have to call discoverDescriptors on the characteristic. 

However, in noble it's not necessary to do ths manually - you could simply call `subscribe()` on your characteristic and grab the incoming 'data' events.

See also [https://www.bluetooth.com/specifications/gatt/descriptors](https://www.bluetooth.com/specifications/gatt/descriptors)

sandeepmistry, 2017-03-14:
Thanks @mattgol!

@prabhakarlad please see @mattgol's answer above.

closed 2017-03-14 by sandeepmistry

prabhakarlad, 2017-03-14:
Thanks @mattgol  @sandeepmistry  I was able to get the descriptor.