noble/noble #567: HRM disconnects after 2 readings

prabhakarlad, 2017-03-14:
I have the following js file which connects to HRM, I see that the it connects to it properly and after 2 callbacks from handleHRMData() it disconnects automatically.  any pointers on this would be helpful.
I have tested the HRM with gattool and it works fine and stays connected.

var noble = require('noble');
var remoteServiceUuid = '180d';

var charuuid = '2a37';
var HRMChar = null;
var srvuuid = '180d';

/* Scan for the custom service ID used by the remote */
noble.on('stateChange', function(state) {
	if (state == 'poweredOn') {
		noble.startScanning([remoteServiceUuid]);
	} else {
		noble.stopScanning();
	}
});

noble.on('discover', function(peripheral) {
	noble.stopScanning();

	console.log('Found '+peripheral.advertisement.localName+' at '+peripheral.address);

	peripheral.connect(function(err) {
		if (err) {
			console.log('Could not connect: '+err);
			return;
		}

		console.log ('Connected');

		founHRMNotify = false; 
		peripheral.discoverServices([remoteServiceUuid], function (err, services) {
			services.forEach(function(service) {
				console.log ('Found service: '+service.uuid);
				service.discoverCharacteristics([], function(err, characteristics) {
					characteristics.forEach (function(characteristic) {
						console.log ('Found characteristic: '+characteristic.uuid);

						if (charuuid == characteristic.uuid) {
							HRMChar = characteristic;
							HRMChar.notify(true, function(error) {});
							HRMChar.on('data', handleHRMData);
						}
					});
				});
			});
		});
		noble.startScanning([remoteServiceUuid]);
	});

	peripheral.once('disconnect', function() {
		console.log ("Disconnected");
		noble.startScanning([remoteServiceUuid]);
	});
});

function handleHRMData (data, isNotification) {
	var rate = data.readUInt8(0);
	var n = (data.length - 2) / 2;
	var rr = [];
	for (var i = 0; i < n; i++) {
		rr.push((data.readUInt16LE(2 + 2 * i) / 1024.0) * 1000.0);
	}
	console.log('hrm ' + rate + 'len: ' + data.length);
}

sandeepmistry, 2017-03-26:
@prabhakarlad please provide an HCI dump capture: `sudo hcidump -t -x`

sandeepmistry, 2017-04-17:
Closing this for now due to lack of feedback.

closed 2017-04-17 by sandeepmistry