noble/noble #568: Could not start scanning, state is: unsupported(not poweredOn)

moshemarciano, 2017-03-15:
Hey Guys,

Noble works pretty well for me on my 2013 macbook pro (dev machine), but when I tried to run it on one of my other macs, an older model iMac and a 2011 macbook pro I end up getting "Could not start scanning, state is: unsupported (not poweredOn)"

state never changes to poweredOn, the only stateChange event I got is "unsupported".

All macs have bluetooth working and enabled and see other devices.

Any ideas?

mainto, 2017-03-18:
I had a same issue. It seems like your mac is not support BLE. You can check on System Report -> Hardware -> Bluetooth. If you can see "Bluetooth Low Energy Supported: No", it won't be working.

moshemarciano, 2017-03-20:
You are right, I bought some USB dongles and now it get picked up but they do behave differently than an on board one.

for example I don't get the localName as I do with the builtin bluetooth chip

sandeepmistry, 2017-03-27:
Closing this, as it was caused by unsupported h/w as `noble.state` reported.

closed 2017-03-27 by sandeepmistry