noble/noble #569: On scanStop, state is undefined when running in pm2

Bonitis, 2017-03-16:
I am trying to set up my node app with noble to run with pm2 so it will start up on boot and and restart if it crashes. When running outside of pm2, I get no errors and everything seems to work fine. When running inside pm2, I get the following debug lines 60+ times per second:

`DEBUG Detected bluetooth scan stop - scanning again`
`DEBUG ScanStop state is: undefined`

Those lines are printed out in this function:

``` javascript
noble.on('scanStop', (state) => {
      log.debug('Detected bluetooth scan stop - scanning again');
      log.debug(`ScanStop state is: ${state}`);
      noble.startScanning(Constants.SERVICE_UUIDS, false, (err) => {
        if (err) {
          log.error(`startScanning error [2]: ${err}`);
        }
      });
    });
```

Despite this, I am still connecting to devices normally and reading information about once per second.

I'm trying to debug why the `scanStop` event would be firing, and also why the state would be undefined. Do you know why that may happen, or how I can best debug this case?

noble: v1.8.0
node: v6.10.0

Bonitis, 2017-03-16:
It looks like I misread some of the docs (my bad!), and there was never a state in the callback from `.on('scanStop')`. That still doesn't explain why that event is firing so frequently when running in pm2, but it at least eliminates one question!

Bonitis, 2017-03-21:
After doing more testing, this doesn't seem to be an issue with pm2, but more a problem that pops up after multiple app restarts or reboots. It is worth noting that I'm running this on a Raspberry Pi 3 with Raspbian.

I haven't nailed down the exact cause, but if I am restarting my node app with noble multiple times, or rebooting the pi, I end up with the situation where the `scanStop` event is firing 60 times per second.

Bonitis, 2017-03-23:
**Update** Since this does not appear to be a pm2 issue, I changed the issue title. The problem occurs after multiple reboots or restarts, but this condition is hard to force, so it has been difficult to track or debug.

sandeepmistry, 2017-03-27:
What's `pm2`?

Bonitis, 2017-03-27:
pm2 is a process manager that will restart a node app after it has crashed. http://pm2.keymetrics.io/docs/usage/quick-start/

I am using it to restart my BLE app on a raspberry pi when power cycling.

Bonitis, 2017-03-30:
Here is an hcidump while this issue is happening: https://gist.github.com/Bonitis/d258d11379934d6f703610f298ee1025

sandeepmistry, 2017-04-10:
@Bonitis is `pm2` started after the `bluetoothd` daemon is started?