noble/noble #570: No compatible USB Bluetooth 4.0 device found

AzertyUser, 2017-03-16:
Keep getting this error. But I do find the device in bluetoothsection of windows. I'm using a usb dongle. 

[image]

ericblade, 2017-03-22:
Also running into this on a PC that has no Bluetooth adapter -- lack of functionality probably shouldn't cause a throw inside a require.  I can't think of any other library I've ever seen that throws while it's being loaded.  Of course, it can easily be worked around, but

`try {
    noble = require("noble");
} catch(err) {
    noble = {
        on: () => {},
    };
}`

is particularly ugly.  I would expect that it might warn, not throw, and just not do anything.

sandeepmistry, 2017-03-27:
@AzertyUser please see: https://github.com/sandeepmistry/noble#windows

@ericblade feel free to PR a suggestion to change the behaviour.

closed 2017-03-27 by sandeepmistry

jayalfredprufrock, 2017-04-20:
I have to agree with @ericblade . I was surprised to find my program completely crash just because a particular computer hadn't gone through the complicated usb bluetooth adapter install process. I thought I was surely missing something until I discovered this issue. What happens on OSX if the computer doesn't have a bluetooth adapter? I assumed that the "poweredOn" state change event just never fires, but now I'm not so sure...