noble/noble #571: Noble for Android?

rei-vilo, 2017-03-16:
I wanted to use BLE with Node-RED on an Android tablet, but Noble isn't available for the Android OS.

[image]

Any plan to add Android to the supported OSes?

Thank you!

sandeepmistry, 2017-03-26:
Hey @rei-vilo,

If make a charge like this: https://github.com/sandeepmistry/bleno/commit/02a1828fa970174900a8cc36298349609db6bb7f to noble and test it out. I'd be happy to merge it :)

rei-vilo, 2017-03-27:
Hello @sandeepmistry

Thank you for the hint. I downloaded the package, updated the files, built and installed it. 

```
$ npm pack noble
$ unzip package
$ npm build package/
$ npm install package/

> usb@1.2.0 install /data/data/com.termux/files/home/.node-red/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v48-android-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@6.10.0 (node-v48 ABI) (falling back to source compile with node-gyp)
gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at failNoPython (/data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:449:14)
gyp ERR! stack     at /data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:353:11
gyp ERR! stack     at F (/data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/which/which.js:69:16)
gyp ERR! stack     at E (/data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/which/which.js:81:29)
gyp ERR! stack     at /data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/which/which.js:90:16
gyp ERR! stack     at /data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/which/node_modules/isexe/index.js:44:5
gyp ERR! stack     at /data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/which/node_modules/isexe/access.js:8:5
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:123:15)
gyp ERR! System Linux 3.10.54
gyp ERR! command "/data/data/com.termux/files/usr/bin/node" "/data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/data/data/com.termux/files/home/.node-red/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/data/data/com.termux/files/home/.node-red/node_modules/usb/src/binding"
gyp ERR! cwd /data/data/com.termux/files/home/.node-red/node_modules/usb
gyp ERR! node -v v6.10.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/data/data/com.termux/files/usr/bin/node /data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/data/data/com.termux/files/home/.node-red/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/data/data/com.termux/files/home/.node-red/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/data/data/com.termux/files/home/.node-red/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 3.10.54
node-pre-gyp ERR! command "/data/data/com.termux/files/usr/bin/node" "/data/data/com.termux/files/home/.node-red/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /data/data/com.termux/files/home/.node-red/node_modules/usb
node-pre-gyp ERR! node -v v6.10.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! not ok
Failed to execute '/data/data/com.termux/files/usr/bin/node /data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/data/data/com.termux/files/home/.node-red/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/data/data/com.termux/files/home/.node-red/node_modules/usb/src/binding' (1)

> bluetooth-hci-socket@0.5.1 install /data/data/com.termux/files/home/.node-red/node_modules/bluetooth-hci-socket
> node-gyp rebuild

gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at failNoPython (/data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:449:14)
gyp ERR! stack     at /data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:353:11
gyp ERR! stack     at F (/data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/which/which.js:69:16)
gyp ERR! stack     at E (/data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/which/which.js:81:29)
gyp ERR! stack     at /data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/which/which.js:90:16
gyp ERR! stack     at /data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/which/node_modules/isexe/index.js:44:5
gyp ERR! stack     at /data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/which/node_modules/isexe/access.js:8:5
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:123:15)
gyp ERR! System Linux 3.10.54
gyp ERR! command "/data/data/com.termux/files/usr/bin/node" "/data/data/com.termux/files/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /data/data/com.termux/files/home/.node-red/node_modules/bluetooth-hci-socket
gyp ERR! node -v v6.10.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
[... cut ...]
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
```

However, although `noble` is listed among the installed nodes, 

```
$ npm list | grep ble
├─┬ noble@1.8.0
```

the `scan BLE` node doesn't appear on the palette.

[image]

Am I doing something wrong on the whole process?

sandeepmistry, 2017-04-09:
@rei-vilo you'll need the `node-gyp` dependencies for it to work: https://github.com/nodejs/node-gyp#installation

Looks like you're missing Python so far ...

rei-vilo, 2017-04-21:
Thank you for the tip and the pointer. I managed to install `python2`, `make`, `g++` and `gcc`. 

Unfortunately, some dependencies of Noble fail to compile, like `libudev`, `node-pre-gyp`. 

Ultimately, Noble still appears as not registered.

[image]

Here is the [trace of my tests](https://gist.github.com/rei-vilo/70f5fd3fac5881e96c0ddcae5e469f93).

sandeepmistry, 2017-04-23:
@rei-vilo you are pretty close, are you sure you've changed this line to add a check for Android?

https://github.com/sandeepmistry/noble/blob/master/lib/resolve-bindings.js#L12

rei-vilo, 2017-04-25:
Yes, and building `udev-package` now complains about missing `node-pre-gyp`, although it is installed. I'm afraid I'm going to give up.

sandeepmistry, 2017-05-08:
@rei-vilo ok, I'm closing this for now.

Btw, you don't need `udev` from npm, the read me refers to the Ubuntu package. Which is only needed for the optional node-usb dependency which is only needed for Windows.

closed 2017-05-08 by sandeepmistry

claudiobadii, 2018-07-02:
Hi, I am trying to install Noble on Android to work with node-red. Do you have idea how to reach the goal?

Now I am blocked here:
[image]

henfri, 2018-10-14:
Hello
apt install libudev-dev could help.
@sandeepmistry  Any chance to get noble installable via npm?

Regards,
Hendrik

henfri, 2018-10-14:
libudev-dev is not available in termux. So, it does not work for me

katonage, 2018-12-28:
Pull request from katonage/noble was made
In termux, installation was fine afterwards, also dependent module like 
npm install node-poweredup
was installed ok ( module for Bluetooth communication with Lego powdered up toys)
To install bluetooth-hci-socket
I installed first:
pkg i python2
npm i gcc
npm i make
npm install -g node-gyp
npm i bluetooth-hci-socket
Than it made the compilation in termux.

Still, functionality was not tested afterwards: I'm not sure how to do it.

Samleo8, 2019-01-24:
I was able to "get" noble by downloading it from a **computer** using `npm i -g noble --save` then using android file transfer/windows file explorer to upload it onto the folder where you need this dependency (connected via USB). Still cannot get the bluetooth functionality working tho

Samleo8, 2019-01-24:
> Pull request from katonage/noble was made
> In termux, installation was fine afterwards, also dependent module like
> npm install node-poweredup
> was installed ok ( module for Bluetooth communication [...]

Were you able to successfully compile an example from `node-poweredup` source tho and get it working on your phone?