noble/noble #573: How to program the pairing of Raspberry Pi and a wearable device in BLE

yongnan0809, 2017-03-17:
We are working on a project to design our IoT gateway based on the Intel Edison or Raspberry Pi (central)
to connect to one of a few wearable devices (peripheral) via BLE. 
We are currently planning to use Noble (Node.js BLE) for this project.
We have to admit that we are beginners in BLE coding.

We have found that one of the current wearable devices requires pairing for BLE communication. 
We can't find example code to do this kind of operation. 

Would you please tell us how to do pairing in Noble? 
If we can get code examples, then that would be great.
We would appreciate your help very much.

Thank you very much

sandeepmistry, 2017-04-17:
Hi @yongnan0809,

Pairing support in noble is somewhat experimental. It only supports "just works" pairing (no pin), and requires the device to respond with `ATT_ECODE_AUTHORIZATION` or `ATT_ECODE_INSUFF_ENC` during a characteristic read or write operation. After this error code is received it will kick of the pairing process transparently.

closed 2017-04-17 by sandeepmistry

yongnan0809, 2017-04-21:
@sandeepmistry 
Thanks for kind explanation.