noble/noble #574: advertisment.localName is undefined on Windows, works on MacOS

moshemarciano, 2017-03-17:
Hi

I am running my code on same location using macOS on one machine and Windows 10 on another with a dedicated speedLink BT4 USB adapter and used Zadig to install WinUSB 6.

Windows 10 running my node code on top of noble detects the same BT devices as my mac with one exception, while my mac usually gets the localName right, the windows machine reports all with "undefined".

I tried looking at noble.js and print out what the discovery method uses to populate the peripheral object and it too was undefined.

did I forget a step in the environment config under windows?

moshemarciano, 2017-03-20:
turns out it does report the *some* of the names *sometimes* and after a long while, I do however always get the address (as opposed to OSX where I only get the uuid)

sandeepmistry, 2017-03-27:
@moshemarciano could please provide some specific details?

Also, debug logs, on Windows:

```
setenv DEBUG=*
node <blah>.js
```

on Mac:

```
DEBUG=* node <blah>.js
```

moshemarciano, 2017-03-27:
I believe the root cause is that even if the device is on standby it still reports its BT presence, however on my mac I immediately get the localName and in windows I only get it when the device is powered on, until that point I only get an address name.

Also, it seems that some devices are not visible to windows at all, just to my Mac.

Finally, my iPhone which reports a random address, also reports it's name, but the name never shows on windows, just on mac.

I tried to debug it but it seems the name never gets in

Mac log:

```
  noble addressChange 3c:15:c2:bc:4a:95 +0ms
  yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1490589961973",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
} +3ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
} +3ms
  yosemite-bindings state change poweredOn +0ms
  noble stateChange poweredOn +0ms
===================> state changed: poweredOn
  yosemite-bindings sendCBMsg: 29, {
  "kCBMsgArgOptions": {
    "kCBScanOptionAllowDuplicates": 1
  },
  "kCBMsgArgUUIDs": []
} +2ms
  noble scanStart +0ms
scan started
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 37,
  "kCBMsgArgs": {
    "kCBMsgArgAdvertisementData": {
      "kCBAdvDataManufacturerData": {
        "type": "Buffer",
        "data": [
          117,
          0,
          66,
          4,
          1,
          128,
          0,
          188,
          20,
          133,
          200,
          60,
          25,
          190,
          20,
          133,
          168,
          60,
          98,
          1,
          0,
          0,
          0,
          0,
          0,
          0
        ]
[... 667 lines cut ...]
  hci           LE meta event data = 0300193cc88514bc1c1bff75004204018000bc1485c83c19be1485a83c6201000000000000d4 +16ms
  hci                   type = 3 +0ms
  hci                   address = bc:14:85:c8:3c:19 +0ms
  hci                   address type = public +0ms
  hci                   eir = 1bff75004204018000bc1485c83c19be1485a83c6201000000000000 +0ms
  hci                   rssi = -44 +0ms
  gap advertisement = {"manufacturerData":{"type":"Buffer","data":[117,0,66,4,1,128,0,188,20,133,200,60,25,190,20,133,168,60,98,1,0,0,0,0,0,0]},"serviceData":[],"serviceUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +16ms
=================> new device detected : undefined | bc:14:85:c8:3c:19
{ id: 'bc1485c83c19',
  uuid: 'bc1485c83c19',
  address: 'bc:14:85:c8:3c:19',
  addressType: 'public',
  connectable: false,
  advertisement:
   { localName: undefined,
     txPowerLevel: undefined,
     manufacturerData: <Buffer 75 00 42 04 01 80 00 bc 14 85 c8 3c 19 be 14 85 a8 3c 62 01 00 00 00 00 00 00>,
     serviceData: [],
     serviceUuids: [],
     solicitationServiceUuids: [],
     serviceSolicitationUuids: [] },
  rssi: -44,
  services: null,
  state: 'disconnected' }
```

sandeepmistry, 2017-04-17:
So the device is not actually advertising a local name:

```
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 37,
  "kCBMsgArgs": {
    "kCBMsgArgAdvertisementData": {
      "kCBAdvDataManufacturerData": {
        "type": "Buffer",
        "data": [
          117,
          0,
          66,
          4,
          1,
          128,
          0,
          188,
          20,
          133,
          202,
          115,
          81,
          190,
          20,
          133,
          172,
          189,
          164,
          1,
          0,
          0,
          0,
          0,
          0,
          0
        ]
      },
      "kCBAdvDataIsConnectable": 0
    },
    "kCBMsgArgRssi": -86,
    "kCBMsgArgDeviceUUID": {
      "type": "Buffer",
      "data": [
        15,
        128,
        95,
        30,
        102,
        64,
        79,
        60,
        183,
        124,
        187,
        41,
        12,
        25,
        233,
        126
      ]
    },
    "kCBMsgArgName": "[TV] UA50J5500"
  }
```

On Mac, the name is retrieved `kCBMsgArgName` is used to populate the local name field ( which is very confusing).

If you look at what the EIR data the device advertises, `bff75004204018000bc1485c83c19be1485a83c6201000000000000` there is only a manufacturer data field and no local name.

I'll close this for now, but am open to changing the Mac code to ignore the `kCBMsgArgName` field ...

closed 2017-04-17 by sandeepmistry