noble/noble #575: bluez version 5.44 compatibility

kdsoo, 2017-03-20:
I'm running noble on arch Linux and since I updated whole arch packages recently I found something weird.

My service built on noble module doesn't work properly. no error but cannot approach my bluetooth dongle which is from broadcom.

Figuring out my last arch linux update, it includes bluez upgrade to version 5.44 and this version has a significant change in bluetooth control.

hci things disappeared. like hciconfig hcidump... bluez claims that they are deprecated and moved on to btmgmt something...which has poor documentation.

I just fixed by rolling back to bluez 5.43 but it must be cleared or needs some heads up in noble documentation.

sandeepmistry, 2017-03-27:
@kdsoo what kind of issues are you seeing?

Any chance you can get an HCI level trace? Maybe `btmon? instead of `hcidump?

kdsoo, 2017-03-27:
I didn't give a try with hci level trace but it obviously could not even scan any ble node at all with noble stack.
It just looked like my linux box has a bluetooth interface but turned off and noble is not giving any error.
Problem is this btmgmt thing has poor documentation so I didn't have much time troubleshooting it.

sandeepmistry, 2017-03-27:
@kdsoo ok, `btmon` or `hcidump` like trace is the best way to figure out what's happening