noble/noble #576: USB Bluetooth adapter reports new device after airplane mode

moshemarciano, 2017-03-20:
Has anyone encountered this?

both on mac and windows when using a USB bluetooth (builtin one doesn't not behave like that)
I get a new device with a new uuid and address (on windows) everytime I switch my iPhone to airplane mode and back...

I am trying to figure out a way to understand and mitigate but so far the only piece of data that is equal on two separate airplane on/off instances is the Manufacturer data..

any thoughts?

moshemarciano, 2017-03-20:
I believe this is because the address type is 'random'

I get public addresses from computers/mac but random address from my iPhone, any idea on how to get the public address of a smartphone without pairing it first?

sandeepmistry, 2017-03-27:
> I believe this is because the address type is 'random'

Correct

> I get public addresses from computers/mac but random address from my iPhone, any idea on how to get the public address of a smartphone without pairing it first?

Nope, privacy feature of iOS :)

closed 2017-03-27 by sandeepmistry