noble/noble #579: Is there noble npm package for anuglar2 implementation?

narada85, 2017-03-22:
I'm trying to integrate noble with angular2 app. but still couldn't bind events.

jacobrosenthal, 2017-03-22:
Theres webbluetooth now, mostly just supported by chrome, and there is a binding from @monteslu that maps that api to the noble api
https://github.com/sandeepmistry/noble/tree/master/lib/webbluetooth
He has a demo running https://noble.surge.sh/

Thats normally packaged for the front end via Webpack or browserify, but I have no idea for angular.

Or you can use the websocket binding people have used on say raspberry pis most commonly to send the bluetooth apis to a separately running node script
https://github.com/sandeepmistry/noble/tree/master/lib/websocket

closed 2017-03-22 by jacobrosenthal

narada85, 2017-03-22:
@jacobrosenthal, thanks for the reply!