noble/noble #582: State change to 'poweredOn' not detected when reseting adapter

Bonitis, 2017-03-28:
**Setup**
noble: v1.8.0
node: v6.10.0
Raspbian Jessie on Raspberry Pi 2
USB BLE antenna

**Adapter**
```
       Type: BR/EDR  Bus: USB
	BD Address: 00:25:BF:70:02:E8  ACL MTU: 310:12  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:1253 acl:0 sco:0 events:69 errors:0
	TX bytes:1066 acl:0 sco:0 commands:65 errors:0
	Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: SLAVE ACCEPT 
	Name: 'raspberrypi-2'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 4.1 (0x7)  Revision: 0x2933
	LMP Version: 4.1 (0x7)  Subversion: 0x2933
	Manufacturer: Cambridge Silicon Radio (10)
```

**Issue**
If I unplug my usb adapter and plug it back in, or run `hciconfig hci0 reset`, while my app is running, it will detect that the adapter state is not `poweredOn` but will not detect when it comes back. If I start the app with the adapter plugged in, it is detected with no issue.

When it first runs, I will get the `Bluetooth Powered On` log. When I unplug it, I get `BLE Antenna state currently 'poweredOff'. Waiting for device to Power on.`. I plug it back in and nothing happens, even though `hciconfig` shows it up.

**Code**
``` javascript
import noble from 'noble';
import Log from 'log';
import colors from 'colors';

const log = new Log('debug');

noble.on('stateChange', (state) => {
   if (state === 'poweredOn') {
     log.debug('Bluetooth Powered On, Scanning for devices...'.green);
   } else {
     log.debug('BLE Antenna state currently', state, '.  Waiting for device to Power on.');
   }
});

noble.on('scanStop', () => {
   log.debug('Detected bluetooth scan stop - scanning again');
});

noble.on('warning', (message) => {
   log.warning(`noble warning: ${message}`);
});
```

sandeepmistry, 2017-04-10:
> When it first runs, I will get the Bluetooth Powered On log. When I unplug it, I get BLE Antenna state currently 'poweredOff'. Waiting for device to Power on.. I plug it back in and nothing happens, even though hciconfig shows it up

Just wondering why you are doing this?

Probably a low priority to fix this ...

Bonitis, 2017-04-10:
I'm doing this because I don't necessarily have control of the device, as they are in the field. I'm trying to account for anything that might happen when another user handles the pi + adapter. One case is someone unplugging it, either on purpose or by accident, and plugging it back in. Right now, it doesn't seem that I have a way to recover my scanning process in that situation.

sathish25071992, 2017-09-11:
Hi,

I am facing the same problem. I am able to detect the dongle disconnect / Bluetooth got poweroff when application was running. But when the dongle is connected back, I am not getting the 'poweredOn' state change. I have to restart my application to make it work again. But the HCI checking device up status every 1 second, Why I am not getting the 'poweredOn'?. Is there any config I can give to make it work? Thank you.

jmhrrrptvj, 2020-01-19:
Hi for me noble state check callback returns only poweredOn even when bluetooth is off ?

mindmakersdisk, 2020-01-31:
I have the same problem running on Raspberry Pi in many schools. Sometime students turned bluetooth adaptar off and on again, but in these case they have to reboot theirs computers because we can't (using noble) to restore our local service to powerOn state. Is there any news about this bug?