noble/noble #583: i want to use more than one BLE devices

yongtingwang, 2017-03-29:
HI

i use two noble at the same time,

var noble1 = require('noble'); 
var noble2 = require('noble'); 

and

noble1.on('stateChange', function(state) {
  console.log('1 on -> stateChange: ' + state);

  if (state === 'poweredOn') {
    noble1.startScanning();      //any service UUID, no duplicates
  } else {
    noble1.stopScanning();
  }
});

noble2.on('stateChange', function(state) {
  console.log('1 on -> stateChange: ' + state);

  if (state === 'poweredOn') {
    noble2.startScanning();      //any service UUID, no duplicates
  } else {
    noble2.stopScanning();
  }
});

in my notebook,it can receive two device's data.
but in my computer, it cannot work.

it will show that 
1 on -> stateChange: poweredOn
1 on -> stateChange: poweredOn
and stop.

what should i do ?

i am not sure about the connection between them.
In my notebook , i use 'read' to get notifications, but in my computer i need to use 'data'

mbifulco, 2017-03-30:
@yongtingwang are you trying to use 2 different bluetooth antennas? There's a note about that sort of use case in the [docs here](https://github.com/sandeepmistry/noble#multiple-adapters).  It looks like you'll need to run two separate node processes.  You may be able to modify this library [here-ish](https://github.com/sandeepmistry/noble/blob/97cf4414fee88d1f29ee7d6aeb2f35fc0613e606/lib/hci-socket/hci.js#L93) to add a parameter to let you specifty which BLE antenna you want to use when you include noble.

yongtingwang, 2017-04-08:
@mbifulco  thank you! i will take a look.

sandeepmistry, 2017-04-10:
@mbifulco thanks!

Closing as answered.

closed 2017-04-10 by sandeepmistry