noble/noble #594: Not scanning without parallel hcitool lescan command

Cosmopal, 2017-04-10:
I have used the following script:

`var noble = require('noble');
console.log('started');	

noble.state="poweredOn";

// state = <"unknown" | "resetting" | "unsupported" | "unauthorized" | "poweredOff" | "poweredOn">

// noble.on('stateChange', callback(state));

noble.startScanning([], true);

noble.on('discover', function(peripheral) { 

  var macAddress = peripheral.uuid;
  var rss = peripheral.rssi;
  var tx = peripheral.txPowerLevel;
  var mandata = peripheral.serviceData;
  var connecteble = peripheral.connectable;
  // process.stdout.write(rss);
  // echo rss;
  // var localName = advertisement.localName; 
  console.log("Hello World!");
  console.log('found device: ', macAddress, ' ', rss,' ', tx,' ',mandata,' ',connecteble);
  if (connecteble=='true'){
  	peripheral.on('connect', function() {
  		console.log('connecting');
            console.log(peripheral.txPowerLevel, 'connected');
        });
  }
});

`
The problem is that the function does not start scanning the ble beacons until I enter `sudo hcitool lescan` in a parallel terminal

Cosmopal, 2017-04-10:
I tried writing a bash script, launching the script first and then doing hcitool lescan, but then it got me only one beacon.

sandeepmistry, 2017-04-17:
I've corrected this in https://github.com/sandeepmistry/noble/commit/7e96ad519ccc146d394c833d2eb7d3dd012c0405. Please try with `npm install sandeepmistry/noble` and report your feedback.

closed 2017-04-17 by sandeepmistry