noble/noble #596: Reporting of beacons not consistent with beacon advertisment interval

camelphd, 2017-04-17:
Hello,

I'm trying to detect beacons with a Raspberry Pi 3 model B, which has a internal Bluetooth antenna. I've got everything installed and I am very impressed what I'm able to do so far.

I have some Eddystone beacons, which transmit with an advertisement interval of 200 ms. I want this high frequency advertisement to have a real-time tracking of the beacon's signal strength.

I slightly altered enter-exit.js reports me the difference in time between each discovery:

```
noble.on('discover', function(peripheral) {

  var id = peripheral.id;
  var entered = !inRange[id];

  if (entered) {
    inRange[id] = {
      peripheral: peripheral
    };
    var time = Date.now()-inRange[id].lastSeen;
    console.log('"' + peripheral.id + '" seen with (RSSI ' +          peripheral.rssi + ') ' + time);

  inRange[id].lastSeen = Date.now();
});
```

Now in the begin, the script reports the beacon each advertisment interval:

```
"d69852fdc1b9" seen with (RSSI -30) 199
"d69852fdc1b9" seen with (RSSI -30) 202
"d69852fdc1b9" seen with (RSSI -46) 200
"d69852fdc1b9" seen with (RSSI -35) 201
"d69852fdc1b9" seen with (RSSI -35) 214
"d69852fdc1b9" seen with (RSSI -41) 24
"d69852fdc1b9" seen with (RSSI -31) 178
"d69852fdc1b9" seen with (RSSI -31) 200
```
However, after 10s, suddenly the script only reports the beacon a few seconds apart. It never returns to fast reporting. For my application this is not acceptable. After a while, the beacon is even not discovered anymore.
```

"d69852fdc1b9" seen with (RSSI -62) 2004
"d69852fdc1b9" seen with (RSSI -54) 6009
"d69852fdc1b9" seen with (RSSI -62) 2004
"d69852fdc1b9" seen with (RSSI -54) 4025
"d69852fdc1b9" seen with (RSSI -54) 1992
"d69852fdc1b9" seen with (RSSI -54) 11879
```

The problem solves itself If I restart the script. But this way of solving is not acceptable.
Do you know any way to solve this? Is it caused the raspberry pi hardware itself?

Thanks for your hard work on Noble.

sandeepmistry, 2017-04-23:
@camelphd can you please share the full Node.js code? As well as an HCI dump capture: `sudo hcidump -t -x`.

Which version of noble are you using?

sandeepmistry, 2017-09-05:
Closing due to lack of activity. Please re-open with requested info.

closed 2017-09-05 by sandeepmistry