noble/noble #598: Stops discovering devices after a few time

asneyap, 2017-04-18:
Hi,

I'm using noble on Xenial 16.04/ bluez 5.43, as well as on Xenial 16.04 / bluez 5.37.  Both of them on an Intel NUC5PPYH mini pc (Intel® Wireless-AC 3165 M.2 2230). They have to be scanning continuously, until they find certain kind of BLE devices, then connect to that device, exchange some activity data, disconnect and start discovering again. The problem is that it may take long time (from minutes to hours) between finding the right devices, and after some time (random) noble stops detecting any LE devices. Moreover, I need to restart bluetoothd service to start seeing devices again and, sometimes, I need to shutdown (not reboot) the machine because the interfaces power off and it is not possible to recover them through something like "sudo hciconfig hci0 up" (both internal ble interface and external usb dongle).

I've expended several weeks trying to find out what is happening to avoid this situation but I'm getting nuts. Please, does anybody guess what's happening? 

Thanks

sandeepmistry, 2017-04-23:
@asneyap strange, could be an HCI adapter issue.

Could you please share the Node.js code you are using, as well as an HCI dump capture (`sudo hcidump -t -x`)?

jrobeson, 2017-09-27:
@asneyap : without more information I'm going to close this. Feel free to reopen with more information.

closed 2018-01-31 by sandeepmistry