noble/noble #601: Use noble library to connect BLE devices at same time go wrong some times.

yongnan0809, 2017-04-21:
Hi.

I’m running noble at raspberry pi 3 to receive multiple BLE wearable devices movement datas.
It’s working very smooth and nice. 

However its weird when program try connect two peripheral device at same time or nearly same time. Noble will connect both of them but just work(Get data) with one of them. Is noble library use shared value in low level ? Or I should not connect multiple BLE device at same time in code?

Second is disconnect peripheral when write characters to peripheral’s handle the the function not return errors. Is this have solution or functions that throw error in this situation? 

Thank you.

sandeepmistry, 2017-04-23:
@yongnan0809 could you please share the Node.js code you are using along with a HCI dump capture (`sudo hcidump -t -x`)?

yongnan0809, 2017-04-24:
I posted a part of code about BLE operating.

Log is here [https://github.com/yongnan0809/Temp/blob/master/log.txt](url)
Code is here [https://github.com/yongnan0809/Temp/blob/master/code.js](url)

log show the second peripheral is not notify data. 
But first connected peripheral is notify data.

Console log is here
```
Get data from db complete
Discoverd  cc78ab6c8804
connectReady now
Step 2: Emitted connect.
peripheral is connected cc78ab6c8804
Step 3: Emitted get characteristics
peripheral is disconnected cc78ab6c8804
Discoverd  cc78ab6c9604
Discoverd  cc78ab6c8804
connectReady now
Step 2: Emitted connect.
peripheral is connected cc78ab6c9604
Step 3: Emitted get characteristics
discovering characteristics
connectReady now
Step 2: Emitted connect.
peripheral is connected cc78ab6c8804
Step 3: Emitted get characteristics
peripheral is disconnected cc78ab6c8804
Discoverd  cc78ab6c8804
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Skin value notification on
Amb value notification on
Lsm value notification on
connectReady now
Step 2: Emitted connect.
peripheral is connected cc78ab6c8804
Step 3: Emitted get characteristics
discovering characteristics
discovering characteristics
discovering characteristics
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
Config OK!
```

It happen when connect multi device at same time. If connect one buy one manually(one by one power on the BLE device) it's not show this problem.

sandeepmistry, 2017-05-08:
This is the weird event from the HCI adapter level ...

```
2017-04-24 10:59:15.795770 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 65 reason 0x3e
    Reason: Connection Failed to be Established
```

yongnan0809, 2017-05-08:
OK,
This mean when I connect multi device at same time, it possible connect fail.

I will correct device connect logic.

Thank you :)

closed 2017-05-08 by yongnan0809