noble/noble #604: is peripheral.rssi using the golden range?

benjaminhon, 2017-04-25:
Can i confirm that the peripheral.rssi returns the actual rssi rather than using hci_read_rssi which according to this is not the actual rssi?

> hci_read_rssi is probably not what you want. It's not the actual remote RSSI. From the BT spec section describing the HCI_Read_RSSI command:

> The RSSI parameter returns the difference between the measured Received Signal Strength Indication (RSSI) and the limits of the Golden Receive Power Range for a Connection Handle to another BR/EDR Controller. Any positive RSSI value returned by the Controller indicates how many dB the RSSI is above the upper limit, any negative value indicates how many dB the RSSI is below the lower limit. The value zero indicates that the RSSI is inside the Golden Receive Power Range.

jacobrosenthal, 2017-04-25:
Dont know where you got that text from or what platform its talking about. Check out each platforms bindings file to see where its taking its rss from. IE:
https://github.com/sandeepmistry/noble/blob/master/lib/mac/yosemite.js
etc

sandeepmistry, 2017-05-08:
It uses the read RSSI command on non-macOS see: https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js#L388-L403

Please re-open with the info @jacobrosenthal requested above.

closed 2017-05-08 by sandeepmistry