noble/noble #607: Installation issues on Raspberry Pi III

FabianKLS, 2017-04-26:
Hi, I'm trying to install Noble on my raspberry pi.

make: Entering directory '/home/pi/nodejs/bluetooth/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/node_usb.cc:1:
../../nan/nan.h:326:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
usb_bindings.target.mk:93: recipe for target 'Release/obj.target/usb_bindings/src/node_usb.o' failed
make: *** [Release/obj.target/usb_bindings/src/node_usb.o] Error 1
make: Leaving directory '/home/pi/nodejs/bluetooth/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'
Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/nodejs/bluetooth/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/nodejs/bluetooth/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)
make: Entering directory '/home/pi/nodejs/bluetooth/node_modules/noble/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../node_modules/nan/nan.h:326:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;

Any ideas?

cyclonmaster, 2017-04-27:
Why not just use npm install noble

It work fine on Rasp Pi 3

sandeepmistry, 2017-05-08:
Please see: https://github.com/sandeepmistry/noble/issues/341#issuecomment-189717602

closed 2017-05-08 by sandeepmistry