noble/noble #610: Make noble work with JXCore, compiled from source, on Mac OSX

TheMaverickProgrammer, 2017-04-29:
I know there's already a similar issue #308 but I don't have the folders or `jx` command line program that the users in the thread use. For instance I don't have a ` ~/.node-gyp/jxb310` folder to delete before rebuilding the modules in my project.

Recap: JXCore works. Noble works. JXCore + Noble does not work:

```
dyld: lazy symbol binding failed: Symbol not found: __ZN4node6Buffer3NewEPKcm
  Referenced from: /Users/marc/Code/noble/node_modules/xpc-connection/build/Release/binding.node
  Expected in: dynamic lookup

dyld: Symbol not found: __ZN4node6Buffer3NewEPKcm
  Referenced from: /Users/marc/Code/noble/node_modules/xpc-connection/build/Release/binding.node
  Expected in: dynamic lookup

Trace/BPT trap: 5
```

Changing node versions, recompiling, rebuilding, none of those steps worked.

TheMaverickProgrammer, 2017-05-04:
any ideas?

sandeepmistry, 2017-05-08:
@TheMaverickProgrammer please open a new issue here to discuss further: https://github.com/sandeepmistry/node-xpc-connection/issues

closed 2017-05-08 by sandeepmistry

TheMaverickProgrammer, 2017-05-08:
this isn't closed at all