noble/noble #611: BLE Write Characteristics

sahilsheoran, 2017-05-03:
I am using https://github.com/sandeepmistry/noble to get BLE(Bluetooth low energy) device details but encountered the following problem. I am able to Scan BLE devices. When i try to get all characteristics it is also working fine, Reading Characteristics is also working fine but when try writing characteristics for BLE devices getting "null" response. Can anyone help, am stuck with this from last week.

sahilsheoran, 2017-05-04:
[image]
Above screenshot shows the error and following is my code which am using for write characteristics : 
characteristic.write(new Buffer([111111111111111111]), false, function(error)
console.log('Write characteristic value.'+error); 

can anybody help its urgent.

stephg, 2017-05-04:
A `null` error parameter value in the write callback just tells you that there is no error.

sahilsheoran, 2017-05-05:
yes @stephg there is no error and still "write characteristic value" is null. This is what i am asking that am not getting any error still my write characteristic value is null, there must be some write characteristics. Any help ?

sandeepmistry, 2017-05-15:
@sahilsheoran please provide the code you are using and an HCI dump capture: `sudo hcidump -t -x`

To me everything looks good ...

sahilsheoran, 2017-05-15:
@sandeepmistry Below is the code which i am using to get the write characteristics of BLE device but getting "null" value : var async = require('async');
var noble = require('../index');

var peripheralIdOrAddress = process.argv[2].toLowerCase();

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});
noble.on('discover', function(peripheral) {						  
  if (peripheral.id === peripheralIdOrAddress || peripheral.address === peripheralIdOrAddress) {
    noble.stopScanning();

    console.log('peripheral with ID ' + peripheral.id + ' found');
    var advertisement = peripheral.advertisement;

    var localName = advertisement.localName;
    var txPowerLevel = advertisement.txPowerLevel;
    var manufacturerData = advertisement.manufacturerData;
    var serviceData = advertisement.serviceData;
    var serviceUuids = peripheral.serviceUuids;

    if (localName) {
      console.log('  Local Name        = ' + localName);
    }

    if (txPowerLevel) {
      console.log('  TX Power Level    = ' + txPowerLevel);
    }

    if (manufacturerData) {
      console.log('  Manufacturer Data = ' + manufacturerData.toString('hex'));
    }

    if (serviceData) {
      console.log('  Service Data      = ' + serviceData);
    }

    if (serviceUuids) {
      console.log('  Service UUIDs     = ' + serviceUuids);
    }

    console.log();
	writeCharacterstic(peripheral);
  }
});
function writeCharacterstic(peripheral)
{
//WRIRE CHARCTERSTIC OF A PARTICULAR SERVICE
console.log('services and characteristics:');

  peripheral.on('disconnect', function() {
    process.exit(0);
  });

  peripheral.connect(function(error) {
	peripheral.discoverServices(['discoveredService'], function(error, services) {
      var immediateAlertService = services[0];
      console.log('discovered service : Service');
      immediateAlertService.discoverCharacteristics(['CHARCTERSTIC'], function(error, characteristics) {
        var alertLevelCharacteristic = characteristics[0];
        console.log('discovered characteristic : Characteristics');
		console.log("characteristics :"+characteristics);
        console.log("error :"+error);
        // true if for write without response
        alertLevelCharacteristic.write(new Buffer([111111111111111111]), false, function(error) { //[0x002d]
        console.log('Write characteristic value.'+error);
        });
      });
    });
	});
}

sandeepmistry, 2017-05-29:
```
$ node
> new Buffer([111111111111111111])
<Buffer c0>
```

The buffer you are writing is probably not what you want. Try `new Buffer('111111111111111111', 'hex')` ?

closed 2017-05-29 by sandeepmistry

sahilsheoran, 2017-05-30:
No, its still not working @sandeepmistry, as i tried to read after writing the buffer new Buffer('0x002c', 'hex') ? which you provided but still same response nothing able to write. Below are services and characteristics and i am trying to write for the service who has the "write characteristics" using the code which i given above with "new Buffer('0x002c', 'hex')" but still nothing i can get, Please check.......
**services and characteristics:**
Service UUID : 1801, Service Name : Generic Attribute
Service UUID : 1800, Service Name : Generic Access
Characteristic UUID: 2a00, Characteristic Name Device Name
    properties  read,,,,,read
    value       4d6963726f5241452d4d303331303030363631313100 | 'MicroRAE-M031000
66111 '
Characteristic UUID: 2a01, Characteristic Name Appearance
    properties  read,,,,,read
    value       0002 | ' ☻'
Service UUID : fc2479406e0811e480fc0002a5d5c51b
Characteristic UUID: f833d6c06e0b11e491360002a5d5c51b
    properties  read, notify, indicate,,,,,read,notify,indicate
    value       0000000000000000000000000000000000000000 | '
'
Characteristic UUID: 3d1158406e0b11e4b24f0002a5d5c51b
    properties  write,,,,,write
Characteristic UUID: 4142b9e0a07411e4b3920002a5d5c51b
    properties  read, notify, indicate,,,,,read,notify,indicate
    value       0000000000000000000000000000000000000000 | '
'
Characteristic UUID: b7a8b060a04911e491980002a5d5c51b
    properties  write,,,,,write
Service UUID : 180a, Service Name : Device Information
Characteristic UUID: 2a29, Characteristic Name Manufacturer Name String
    properties  read,,,,,read
    value       52414520627920486f6e657977656c6c000000000000 | 'RAE by Honeywell
      '
Characteristic UUID: 2a24, Characteristic Name Model Number String
    properties  read,,,,,read
    value       50474d2d323630300000000000000000000000000000 | 'PGM-2600
      '
Characteristic UUID: 2a25, Characteristic Name Serial Number String
    properties  read,,,,,read
    value       4d303331303030363631313100000000000000000000 | 'M03100066111
      '
Characteristic UUID: 2a26, Characteristic Name Firmware Revision String
    properties  read,,,,,read
    value       312e303820000000 | '1.08    '
Characteristic UUID: 2a27, Characteristic Name Hardware Revision String
    properties  read,,,,,read
    value       0000000000000000 | '        '
Characteristic UUID: 2a08, Characteristic Name Date Time
    properties  read,,,,,read
$↔' value       e007061b0d241d | '`♠
Characteristic UUID: 2a28, Characteristic Name Software Revision String
    properties  read,,,,,read
    value       0000000000000000 | '        '
Characteristic UUID: 2a23, Characteristic Name System ID
    properties  read,,,,,read
    value       123456ffff9abcde | '↕4V⌂⌂→<^'
Characteristic UUID: 2a50, Characteristic Name PnP ID
    properties  read,,,,,read
    value       ffffffffffffff | '⌂⌂⌂⌂⌂⌂⌂'
Service UUID : 180f, Service Name : Battery Service
Characteristic UUID: 2a19, Characteristic Name Battery Level
    properties  read,,,,,read
    value       5c | '\'
Service UUID : 9e5d1e475c1343a0863582ad38a1386f
Characteristic UUID: e3dd50bff7a74e99838e570a086c666b
    properties  write, notify, indicate,,,,,write,notify,indicate
Characteristic UUID: 92e86c7ad9614091b74f2409e72efe36
    properties  write,,,,,write
Characteristic UUID: 347f76082e2d47eb913b75d4edc4de3b
    properties  read,,,,,read
    value       193a0100 | '↓:☺ '

Its very much urgent now, Help!!!

thangta, 2018-07-23:
Hi, I got same error, Does your problem is fixed?

2krishna1996, 2019-09-12:
hey there. i know it's too late. Does your problem fixed? if not, please feel free to comment. :)