noble/noble #613: Erratic times between scan discovery

joshfarrant, 2017-05-07:
I'm currently using noble on a MacBook running Sierra and trying to discover a Puck.js that's advertising a counter variable to indicate button presses. I never actually connect the Puck.js, I just read the values it's advertising.

I've noticed that there can be quite a range in the amount of time between each `on('discover')` event being emitted. For example, the following (simplified) code.

```js
const noble = require('noble');

noble.on('stateChange', (state) => {
  if (state === 'poweredOn') {
    noble.startScanning([], true);
  } else {
    noble.stopScanning();
  }
});

let lastDate = Date.now();
noble.on('discover', () => {
  const now = Date.now();
  const diff = (now - lastDate) / 1000;
  console.log(`Discovered! Time since last reading: ${diff}s.`);
  lastDate = now;
});

```

Produces the following output:

[image]

Noble seems to be able to discover at most once every 1.5 seconds, but sometimes it can be 20+ seconds between discoveries. Granted, I'm using Noble in quite a hacky way (checking an advertised counter variable to see if the button has been pressed) but I wanted to find out if there was anything I could be doing to make the time between these discoveries be more consistently around the 1.5 second mark.

sandeepmistry, 2017-05-08:
@joshfarrant could you provide some debug logs: `DEBUG=* node <file>.js`

joshfarrant, 2017-05-08:
No problem. Here's a minute of logs.

https://pastebin.com/GbNk169i

Let me know if there's anything else you need.

sandeepmistry, 2017-05-15:
@joshfarrant do the `"kCBMsgId": 37` lines correspond with when a device is discovered and reported in the event?

Are any missing? It would be great if you could annotate the debug log to include any missing events. Thanks.

fafa92, 2017-06-16:
Any results? I have the same problem!

joshfarrant, 2017-06-17:
I switched to using a Raspberry Pi, rather than my Macbook, and everything then started working fine.

Maybe the issue lies with the Macbook itself? It doesn't look like it's a noble issue.

jrobeson, 2017-06-17:
It could also be the library noble uses to talk to mac os bluetooth stack, but I don't use mac os, so I can't be sure.

fafa92, 2017-06-17:
@joshfarrant thanks for your clarification, but I have this problem with my Raspberry Pi zero actually! after few seconds it cannot discover any BLE device.

sandeepmistry, 2017-06-18:
@joshfarrant thanks for the update. It looks like something internal with CoreBluetooth/blued on macOS which is requires Apple's help. I'll close this for now since you have a workaround.

closed 2017-06-18 by sandeepmistry