noble/noble #614: Fails to install via NPM on Raspberry Pi Zero

fafa92, 2017-05-07:
After trying to run command  ` npm install noble`  I got this error:
bluetooth-hci-socket@0.5.1 install /home/pi/noblee/node_modules/noble/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory '/home/pi/noblee/node_modules/noble/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../node_modules/nan/nan.h:326:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
binding.target.mk:82: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/pi/noblee/node_modules/noble/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.50+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/noblee/node_modules/noble/node_modules/bluetooth-hci-socket
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.5.1
noble@1.8.1 node_modules/noble
├── bplist-parser@0.0.6
└── debug@2.2.0 (ms@0.7.1)

Anyone knows what's the problem?

sandeepmistry, 2017-05-08:
@fafa92 please see: https://github.com/sandeepmistry/noble/issues/341#issuecomment-189717602

closed 2017-05-08 by sandeepmistry

fafa92, 2017-05-08:
I updated my node and npm and node-gyp still the same issue:
node -v
v7.10.0
npm -v
1.4.21
node-gyp -v
v0.12.2