noble/noble #619: adapter state unauthorized even after running setcap cap_net_raw+eip

sathish25071992, 2017-05-10:
Hi,

I want to use Bluetooth functionality in my application. I installed noble and im able to launch my application. But Im getting following warning,

> noble warning: adapter state unauthorized, please run as root or with sudo
>                or see README for information on running without root/sudo:
>                https://github.com/sandeepmistry/noble#running-on-linux

Even after giving ca_net_raw privilege to node ( `sudo setcap cap_net_raw+eip $(eval readlink -f `which node`)`), I am getting the same warning. 
Im using noble version 1.8.1

sandeepmistry, 2017-05-14:
@sathish25071992 what does:

```
readlink -f `which node`
```

and

```
sudo getcap $(eval readlink -f `which node`)
```

output?

sathish25071992, 2017-05-25:
Hi sandeep,
Sorry for the late reply. Got stuck with some work. The output of `readlink -f 'which node'` is` /usr/bin/node` and output of `sudo getcap $(eval readlink -f 'which node')` is `/usr/bin/node = cap_net_raw+eip`

sandeepmistry, 2017-05-29:
@sathish25071992 you'll have to run `sudo setcap cap_net_raw+eip <binary>, where `<binary>` is the file for the Electron binary.

closed 2017-05-29 by sandeepmistry

yogendranyadav, 2019-04-04:
Hi Sandeep,
  I still get this issue. This is caused when i tried to use cylon-ble with reference to https://developer.ibm.com/recipes/tutorials/control-your-spherobb8-using-ibm-watson-iot-platform-and-nodered/

I am using node version: 8.11.1
Noble version: 1.9.1
I am running the nodes via Node-red. I tried all the steps mentioned above. It doesn't seem to solve the issue.  

# noble warning: adapter state unauthorized, please run as root or with sudo or see README for information on running without root/sudo: https://github.com/sandeepmistry/noble#running-on-linux
Also tried all the steps as per the above mention link,
Help would be appreciated.

**Outputs:**
pi@raspberrypi:~ $ readlink -f `which node`
/usr/bin/node
pi@raspberrypi:~ $ sudo getcap $(eval readlink -f `which node`)
pi@raspberrypi:~ $