noble/noble #624: Most of the times can't get the services of a device

alvarosp, 2017-05-26:
Hi,
I am running noble in an [Artik 530](https://www.artik.io/modules/artik-530/) that runs Fedora and am trying to access the services and characteristics of an [Artik 020](https://www.artik.io/modules/artik-020/) that runs a Thermometer example app. I can access the information in the Artik 020 when running LightBlue Explorer app from iOS and LightBlue app from a mac and also when using bluetoothctl from the Artik 530 without any problem. But when I try the peripheral-explorer.js example most of the time I get no info:
```
peripheral with ID 000b570be269 found
  Local Name        = BG57961
  Service Data      = 
  Service UUIDs     = 1809,1802

services and characteristics:
```

Only sometimes I get some info:
```
peripheral with ID 000b570be269 found
  Local Name        = BG57961
  Service Data      = 
  Service UUIDs     = 1809,1802

services and characteristics:
1801 (Generic Attribute)
1800 (Generic Access)
180a (Device Information)
1809 (Health Thermometer)
  2a05 (Service Changed)
    properties  indicate
```
And very rarely I get all the information:
```
peripheral with ID 000b570be269 found
  Local Name        = BG57961
  Service Data      = 
  Service UUIDs     = 1809,1802

services and characteristics:
1801 (Generic Attribute)
  2a05 (Service Changed)
    properties  indicate
1800 (Generic Access)
  2a00 (Device Name)
    properties  read, write
    value       42473537393631 | 'BG57961'
  2a01 (Appearance)
    properties  read
    value       0003 | ''
180a (Device Information)
  2a29 (Manufacturer Name String)
    properties  read
    value       53696c69636f6e204c616273 | 'Silicon Labs'
1809 (Health Thermometer)
  2a1c (Temperature Measurement)
    properties  indicate
  2a1d (Temperature Type)
    properties  read
    value       02 | ''
  2a1e (Intermediate Temperature)
    properties  notify
  2a21 (Measurement Interval)
    properties  write
1802 (Immediate Alert)
  2a06 (Alert Level)
    properties  writeWithoutResponse
1d14d6eefd634fa1bfa48f47b42119f0
  f7bf3564fb6d4e5388a45e37e0326063
    properties  write
```
I am running node.js v7.10.0 (tried also v4.7.3 that is the one that came preinstalled but nothing) and noble 1.8.1

I attach the different logs.

Any idea what might be happening?
Thank you.

[bluetoothctl.txt](https://github.com/sandeepmistry/noble/files/1032361/bluetoothctl.txt)
[connection_not_ok.txt](https://github.com/sandeepmistry/noble/files/1032364/connection_not_ok.txt)
[connection_ok.txt](https://github.com/sandeepmistry/noble/files/1032363/connection_ok.txt)
[connection_pseudo_ok.txt](https://github.com/sandeepmistry/noble/files/1032362/connection_pseudo_ok.txt)

jrobeson, 2017-08-05:
Can you replicate this with other devices ? If you have no other devices you could try using the gattserver on the NRF Connect app from Nordic. Or some other computer acting as a peripheral.

sandeepmistry, 2017-09-04:
@jrobeson thanks for helping out!

@alvarosp please re-open with answers to the questions @jrobeson asked above. Thanks.

closed 2017-09-04 by sandeepmistry