noble/noble #625: 128-bit service data advertisement is not parsed.

arekzelechowski, 2017-05-31:
OS: Linux ARM v7
BlueZ v5

Currently if we look at [gap.js](https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/gap.js#L193), we can see that noble parses advertisement/scan response packets to fill `serviceData` property.

According to the [docs](https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile), packets can specify 16-bit (`0x16`), 32-bit (`0x20`) and 128-bit (`0x21`) UUIDs service data. However, noble does use only 16-bit UUID.

On the other hand, noble parses 128-bit service list (`0x06` and `0x07`).

Example:
`02010604160f18401321cf6ea2ffad809e8d36443b5ab56f6d2b0040` is parsed to:
`{"serviceData":[{"uuid":"180f","data":"5d"}]`
Instead of:
`{"serviceData":[{"uuid":"180f","data":"5d"},{"uuid":"2b6d6fb55a3b44368d9e80adffa26ecf","data":"0040"}]`

sandeepmistry, 2017-06-18:
Closed via #626. Thanks @arekzelechowski!

closed 2017-06-18 by sandeepmistry

rzr, 2020-04-30:
Relate-to: https://github.com/abandonware/noble/issues/81#issuecomment-621761981