noble/noble #628: Unreliable signal detection?

kaYcee, 2017-06-02:
Great package! One thing I noticed though is that with a beacon that send at a constant rate (e.g. 500ms), the noble scanner starts working correctly, but after short time it "swallows" some signals only to push them out in a row a couple of seconds later, sometimes the delay is a couple of seconds. Usually that's not a big deal but I am trying to get a realtime detector running and it currently triggers a lot of false alarms over a day. I am using the latest version on RPi3. Thanks!

sandeepmistry, 2017-06-18:
@kaYcee maybe you could share your code and an HCI dump capture (`sudo hcidump -t -x`) along with the debug output.

cruepprich, 2017-08-28:
kaYcee, 
I'm trying to get noble running on my Pi3, but get all kinds of errors. Which OS are you running? Did you have to do anything special to get noble going?

Thanks!

bastianleirpo, 2017-08-31:
We are having the same issue here. Any solutions so far? 
It might be the same problem than reported #596
We are using: Raspberrry Pi 3 b, internal Bluetooth sensor, noble 1.8.1, up-to-date raspberian via noobs, Estimote beacons

sandeepmistry, 2017-09-05:
@bastianleirpo please provide info requested in https://github.com/sandeepmistry/noble/issues/628#issuecomment-309278901 then we can re-open this.

I'm closing it due to lack of activity for now.

closed 2017-09-05 by sandeepmistry