noble/noble #629: Unable to install noble on Win7

Xan-Kun, 2017-06-02:
Hello!
I am running Win7/x64 and started as advised:

1. install **python-2.7.13.msi**
2. install **node-v6.10.3-x64.msi**
3. install **WinUSB** driver with zadig
4. **npm install --global --production windows-build-tools** no problems here, installation finishes with
Waiting for installers... -Successfully installed Visual Studio Build Tools. Successfully installed Python 2.7
C:\Users\aaa\AppData\Roaming\npm `-- windows-build-tools@1.2.1
5. **npm install** in the noble-master directory

I get the following error message:

```
> node-gyp rebuild

c:\dev\noble-master\node_modules\bluetooth-hci-socket>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules
\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )
c:\dev\noble-master\node_modules\bluetooth-hci-socket\build\binding.sln : error MSB3411: Could not load the Visual C++ component "V
CBuild.exe". If the component is not installed, either 1) install the Microsoft Windows SDK for Windows Server 2008 and .NET Framew
ork 3.5, or 2) install Microsoft Visual Studio 2008.
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v3.5\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd c:\dev\noble-master\node_modules\bluetooth-hci-socket
gyp ERR! node -v v6.10.3
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@~0.1.4 (node_modules\xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (cu
rrent: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules\bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
```

I tried setting the msvs_version wihtout success and also tried different versions of gyp and npm but he errors just got worse (until a point where I had to reset the VM and start over).

I am very confused on how to proceed and any hint will be greatly appreciated. I wonder if I am doing something more general wrong. Should I try different python versions or keep on trying npm versions?
As you might have guessed I am very new to npm and node but I am totally willing to help and write about how I got it to work (hopefully). If I can provide any information or try things out, I would gladly do so.

Thanks in advance for the great project!

rasgo-xyz, 2017-06-02:
You need to install node-gyp. A bluetooth-hci-socket prerequisite:

https://github.com/sandeepmistry/node-bluetooth-hci-socket
https://github.com/nodejs/node-gyp#installation

Xan-Kun, 2017-06-02:
A version of node-gyp was already present (I am not sure how) and I tried again from scratch, this time with an extra install node-gyp. It is still the same error message about the VS2008 compiler.

Xan-Kun, 2017-06-02:
I made more trial and error and I think it has something to do with the gyp version, but I can't figure out what's wrong.

Xan-Kun, 2017-06-02:
I THINK I just found the problem; I forgot to install .NET 4.5.1
I will try with a fresh install and see, if that really was all...

Xan-Kun, 2017-06-02:
BTW, I think I forgot to install .NET because I used the video tutorial from Don Coleman and he uses VS, which automatically installs all the .NET stuff and the compiler. I wanted to go without VS and it works just fine :-)

closed 2017-06-03 by Xan-Kun