noble/noble #635: BLE scanning goes slow after few seconds

fafa92, 2017-06-13:
I'm using noble for scanning and detecting close BLE devices, I run advertisement-discovery.js for that, at first it detects devices very fast but after few seconds it takes a lot to detect new devices that have been entered in the area. Has anyone had the same problem here?

jrobeson, 2017-06-17:
You should mention your OS, node versoon, and noble version.

fafa92, 2017-06-17:
It's Raspbian Jessie, 7.7.1 and the latest one

sandeepmistry, 2017-06-18:
@fafa92 could you please share an HCI dump capture (`sudo hcidump -t -x`) along with the Node.js console output. Please also annotate the events you expect or don't see. Thanks.

fafa92, 2017-06-19:
@sandeepmistry I did it and it seems it works good, but I understand the case of my problem since noble library detect devices based on their MAC address, so basically if you set your code to detect every device just once(without duplicates) so when I want to detect my iPhone which is turned to an iBeacon, just when I turn my screen on and unlock it my Raspberry Bluetooth and noble can detect it as device with specific MAC address but since I have to go to my app to set it as an iBeacon and detect its UUID now it cannot detect it anymore. so do you have any solution for that? 
I know I can play with finding duplicates and do my stuff but I just wonder if it's possible to find a way to around that case. Thanks....

elafargue, 2017-07-06:
I have the exact same issue on MacOS (Sierra, Macbook): the first couple of BLE peripherals are discovered right away, then each new peripheral is discovered more and more slowly, up until 10+ seconds between each discovery - using another BLE scanner, I don't see this slowdown on any of the peripherals. 

The consequence is that I often cannot connect to the peripheral I'm looking for, because the scan just doesn't find it 40% of the time because of the slowdown... Other scanners see the same peripheral right away, since it advertises often.

sandeepmistry, 2017-07-31:
@fafa92 

> so when I want to detect my iPhone which is turned to an iBeacon, just when I turn my screen on and unlock it my Raspberry Bluetooth and noble can detect it as device with specific MAC address but [...]

This is a limitation of iOS from my understanding.

@elafargue please open a new issue to track your issue. Thanks.

closed 2017-07-31 by sandeepmistry

OscarCanek, 2019-06-10:
@elafargue it has been a couple of years. I know is iOS but, did you find the real cause of the behaviour?

I'm stuck with the same consequences