noble/noble #636: Memory Leak

duncanhoggan, 2017-06-14:
I've been using noble and bleacon for a while, for recording data from a few hundred BLE beacons broadcasting fairly frequently and have run into a memory leak with running simple tasks.

```
const noble = require('noble')

noble.on('stateChange', function (state) {
	if (state == "poweredOn") {
		noble.startScanning()
	}
})

noble.on('discover', function (perip) {
	console.log(perip.id)
})
```

And after half a day of running i get the following crash.

```
<--- Last few GCs --->
[2535:0x31dd550] 40825511 ms: Mark-sweep 1327.0 (1350.3) -> 1326.9 (1350.3) MB, 2171.0 / 0.0 ms  allocation failure GC in old space requested
[2535:0x31dd550] 40827774 ms: Mark-sweep 1326.9 (1350.3) -> 1326.9 (1348.8) MB, 2262.2 / 0.0 ms  last resort 
[2535:0x31dd550] 40830039 ms: Mark-sweep 1326.9 (1348.8) -> 1326.9 (1348.8) MB, 2265.5 / 0.0 ms  last resort 
<--- JS stacktrace --->

==== JS stack trace =========================================
Security context: 0xa2cb70a66a1 <JS Object>
    1: onDiscover [/home/avatar7/beacon_listener/node_modules/noble/lib/noble.js:~109] [pc=0x12f4b33a57bc](this=0x2b54c19c5ec9 <a Noble with map 0x390337a29299>,uuid=0x2d80e2fbf7f9 <String[12]: f074d2f5287e>,address=0x2d80e2fbf889 <String[17]: f0:74:d2:f5:28:7e>,addressType=0x1636abc12c29 <String[6]: random>,connectable=0x1636abc02421 <false>,advertisement=0x2d80e2fbf841 <an Object...

FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory
 1: node::Abort() [node]
 2: 0x134e91c [node]
 3: v8::Utils::ReportOOMFailure(char const*, bool) [node]
 4: v8::internal::V8::FatalProcessOutOfMemory(char const*, bool) [node]
 5: v8::internal::Factory::NewFixedArray(int, v8::internal::PretenureFlag) [node]
 6: v8::internal::HashTable<v8::internal::NameDictionary, v8::internal::NameDictionaryShape, v8::internal::Handle<v8::internal::Name> >::New(v8::internal::Isolate*, int, v8::internal::MinimumCapacity, v8::internal::PretenureFlag) [node]
 7: v8::internal::Dictionary<v8::internal::NameDictionary, v8::internal::NameDictionaryShape, v8::internal::Handle<v8::internal::Name> >::EnsureCapacity(v8::internal::Handle<v8::internal::NameDictionary>, int, v8::internal::Handle<v8::internal::Name>) [node]
 8: v8::internal::Dictionary<v8::internal::NameDictionary, v8::internal::NameDictionaryShape, v8::internal::Handle<v8::internal::Name> >::Add(v8::internal::Handle<v8::internal::NameDictionary>, v8::internal::Handle<v8::internal::Name>, v8::internal::Handle<v8::internal::Object>, v8::internal::PropertyDetails, int*) [node]
 9: v8::internal::LookupIterator::ApplyTransitionToDataProperty(v8::internal::Handle<v8::internal::JSObject>) [node]
10: v8::internal::Object::AddDataProperty(v8::internal::LookupIterator*, v8::internal::Handle<v8::internal::Object>, v8::internal::PropertyAttributes, v8::internal::Object::ShouldThrow, v8::internal::Object::StoreFromKeyed) [node]
11: v8::internal::Object::SetProperty(v8::internal::LookupIterator*, v8::internal::Handle<v8::internal::Object>, v8::internal::LanguageMode, v8::internal::Object::StoreFromKeyed) [node]
12: v8::internal::Runtime::S
```etObjectProperty(v8::internal::Isolate*, v8::internal::Handle<v8::internal::Object>, v8::internal::Handle<v8::internal::Object>, v8::internal::Handle<v8::internal::Object>, v8::internal::LanguageMode) [node]
13: v8::internal::Runtime_SetProperty(int, v8::internal::Object**, v8::internal::Isolate*) [node]
14: 0x12f4b2c043a7

As a "hack fix" I've been using a cron to manually GC, but it's not a propper solution.
If you want me to grab the heap as it grows let me know.

Thanks for all the great work you've done in the Beacon space.

jrobeson, 2017-06-17:
Which OS, node version, and noble version?

sandeepmistry, 2017-06-18:
@duncanhoggan also what beacons are you using, do they advertise with (rotating) random addresses?

duncanhoggan, 2017-06-20:
@jrobeson 
OS: Ubuntu 16.04 LTS
Node: 7.10.0
Noble: 1.8.1

@sandeepmistry Using the estimote stickers through bleno and noble and yes they appear to be using the rotating addresses. I suspect that that is where the memory is going now that you mention it.

ajitam, 2017-07-17:
Hello,
I'm also experiencing the same thing.

I did some heap snapshots and I can see that "Peripheral" object is constantly growing. I would expect that if I set `duplicates=False` in `startScanning`

```
raspi-0-W
npm: 3.10.10
Node: v6.11.0
```

dstarke, 2017-08-02:
I noticed that here:
https://github.com/sandeepmistry/noble/blob/master/lib/noble.js#L167
The UUID is pushed into an array, but nothing I can see ever removes them. If `allowDuplicates` is true, this should probably be skipped, and if it isn't, there should probably be some sort of cleanup process to remove things that are old.

I'd also recommend using a hash instead of an array for `_discoveredPeripheralUUids`, with the UUID as key and an update time as value, so you can easily go through the keys and remove the old ones.

jrobeson, 2017-08-03:
what do other bluetooth libs do to handle this situation? What heuristic do they use to clean up in the situation of rotating addresses?

dstarke, 2017-08-04:
I'm testing a set of changes that ages out cached peripheral information after a set amount of time. For the moment, I have it set to do this only when `allowDuplicates` is true and you are actively scanning. 

I'll publish information about my fork and submit a pull request after I've tested it a little more.

dstarke, 2017-08-11:
I'm till testing, but I have a preliminary working version here, if anyone wants to look at it or test it out:
https://github.com/Atollogy/noble/tree/memory-leak-fix

In this branch, I've moved all the peripheral caching to a separate class. When `allowDuplicates` is false, the behavior should be the same as at present. When `allowDuplicates` is true, the cache is reset each time you `startScanning()`, and the cache is periodically swept, removing any entries that are not currently connected and have not been accessed or modified in more than 5 seconds.

That behavior should prevent the cache from growing indefinitely when `allowDuplicates` is true.

@duncanhoggan It sounds like you have a test case. If you could point the noble dependency for your test at that branch and re-run your test, that would be very helpful (also, make sure you set `allowDuplicates` to `true` when you start scanning). Here's a line you can copy into the dependencies section of your package.json file to make that easy:
`"noble": "github:atollogy/noble#memory-leak-fix"`

dstarke, 2017-10-10:
So I've been running this a while, and it helps, but it looks like the next layer (Gap) may also be caching advertisements by address. This appears to cause some memory growth over time, although it is slower than before the above changes.

The strategy of caching a previous advertisement and notifying with a mix of new and cached data is pervasive in this library, and can lead to some misleading results. Consider a beacon that sends an advertisement followed by a scan response. This library will notify once with the advertisement data, and then again with a mix of the original advertisement data and the scan response data. When a client receives a discovery notification from Noble, it isn't easy to determine if the advertisement being discovered is in the first category or the second.

The easiest way to fix the memory growth problem would be to not have the lower layers cache anything and just pass events through. If there must be caching and merging of events, centralize the logic for it in one place, such as the noble.js layer where (in my branch, linked above) I've added a cache that is managed and swept. It's probably difficult to change this behavior, though, without potentially breaking existing code.

If anyone has some thoughts about how to go about this, I'd love to hear them. I'm planning to proceed along these lines as I am affected by both the memory growth and double notification bugs.

dstarke, 2017-10-10:
In working through this, I'm assuming that the `advertisement` property of a Peripheral should contain the merged historical advertisement state, since that's sort of what it does currently. My plan is to add a second parameter to Noble's `onDiscover()` event (the first is the Peripheral) that contains just the current advertisement contents, in case a consumer wants to manage their own state.

dstarke, 2017-12-21:
I've added the pull request to address this issue. 

As described above, and in the PR, there's a minor change to the interface for `onDiscover()`, where a second parameter has been added to contain the actual advertisement contents. The new cache behavior should be in effect when `allowDuplicates` is true. When `allowDuplicates` is false, the cache is not swept or cleared, leaving the behavior similar to the way it was before. 

These changes also addressed some inconsistencies in how scan responses were being handled relative to the Bluetooth spec.