noble/noble #637: Beacon needs to power on again to be detected

Gayatrideshmukh, 2017-06-14:
I am able to identify my beacons but only when i power on them. If beacon is already powered on then it does not detect it.

I need to manually power off the beacon and then power on.

I need to keep the beacon status updated every time 
Please let me know if you need more details to understand the problem.

sandeepmistry, 2017-06-18:
@Gayatrideshmukh what code are you using for scanning. I think you will need to enable scanning with duplicates ...

Gayatrideshmukh, 2017-06-19:
Actually i allwed the duplicated but still it was not showing until i power on them.
I did more analysis and found that hcitool.lescan also not showing the device until i power on them.

I found another library 'node-bleacon' which is helping me. its detecting the devices which are already on with few more details like accuaracy , distance in meter etc. but the problm with this library is i am not able to unquely identify devices.
I tried to change in lib of node-bleacon by adding
bleacon.address = peripheral.address and so many things tried in lib also.

Can you please help me to identify the unique feature so that i can differentiate the devices ?
Also one more problem is 'Its not detecting series 10 beacon' . its working with series 11 beacons.

Thanks,
Gayatri

Gayatrideshmukh, 2017-06-19:
Following is the code using noble

var noble = require('noble');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn')
    noble.startScanning([], true); // any service UUID, allow duplicates
  else
    noble.stopScanning();
}); 

noble.on('discover', function(peripheral) { 

  var macAddress = peripheral.uuid;
  var rss = peripheral.rssi;
  
if(macAddress=='a4d856183313' || macAddress =='a4d85618e05e' || macAddress =='a4d8560c110a')
{
  console.log('found device: ', macAddress, ' ', rss);   
}

});

The problem  : need to power on to detect. hcitool.lescan also doing same.

Gayatrideshmukh, 2017-06-20:
Hi Sandeep,

By using noble , we are able to detect the beacons which are already powered on by chanining the code like below.

if (state === 'poweredOn')
{ 
noble.startScanning([], true); // any service UUID, allow duplicates
}
else
{
noble.on('stateChange', function(state) {
noble.startScanning([], true);
});
}

Now we are almost there but the problem is after few seconds device stop detecting.
Its not continuously detecting device. Is it something like device stops advertising , if yes then what we need to do to continuously detect it.

Gayatrideshmukh, 2017-06-22:
Hi ,

My issue is resolved by pushing using the Bleacon library.
By pushing the proper configuration by Gimbal manager , it helped me.
We set unique UUID, major , minor from Gimbal manager for the beacon.
We applied assigned configuration to those beacon.
after pushing those configuration from app store beacon manager, we were getting the uuid which we were specified.

Hope this help someone who will be reading this.
Thanks,
Gayatri

sandeepmistry, 2017-07-30:
Closing this as resolved. @Gayatrideshmukh thanks for the updates.

closed 2017-07-30 by sandeepmistry