noble/noble #642: Noble connects to device multiple times

boazdejong, 2017-06-23:
The first time I scan and connect to my device, everything works as expected. Then when I try again, it connects to the peripheral twice. Then three times, etc.. Is there an explanation for this behaviour?

```
import noble from 'noble';

export function startScanning() {
  return noble.startScanning([], true);
}

export function checkIn(name, callback) {
  return noble.on('discover', (peripheral) => {
    if (peripheral.advertisement.manufacturerData) {
      const mode = Array.prototype.slice.call(peripheral.advertisement.manufacturerData, 0)[4];
      if (mode === 4 && peripheral.rssi > -45) {
        noble.stopScanning();
        peripheral.connect();
        peripheral.once('connect', () => {
          console.log('Connected');
          peripheral.discoverServices('7b00aaaa4be04c349e1e91ec7b084af5');

          peripheral.once('servicesDiscover', (services) => {
            const service = services[0];
            service.discoverCharacteristics('7b00aaa34be04c349e1e91ec7b084af5');
            service.once('characteristicsDiscover', (characteristics) => {
              const characteristic = characteristics[0];
              characteristic.subscribe();
              peripheral.writeHandle(0x0025, new Buffer([0x02, 0x00]));
              peripheral.writeHandle(0x0024, new Buffer('8DF2BFE08299810760EF006EAD486952B2BA38D7', 'hex'));

              characteristic.on('data', (data) => { // eslint-disable-line
                if (data[0] === 0x01) {
                  const str = name;
                  const nameBuffer = Buffer.allocUnsafe(str.length);
                  for (let i = 0; i < str.length; i += 1) {
                    nameBuffer[i] = str.charCodeAt(i);
                  }
                  peripheral.writeHandle(0x001D, nameBuffer);
                } else if (data[0] === 0x02) {
                  return callback({
                    statusCode: 500,
                    message: 'Wrong passkey entry',
                  });
                } else if (data[0] === 0x10) {
                  peripheral.disconnect();
                  return callback({
                    statusCode: 200,
                    message: 'Check in successful',
                    deviceAddress: peripheral.address,
                    info: name,
                  });
                }
              });
            });
          });
        });
      }
    }
  });
}
```

manuelmitasch, 2017-07-17:
How often do you invoke the checkIn function?
If you invoke it multiple times, the discover handler will be registered for each invocation.
This would lead to the behaviour you described.

sandeepmistry, 2017-07-31:
@boazdejong please see @manuelmitasch's comment above and re-open with your answer.

@manuelmitasch thanks for helping out!

closed 2017-07-31 by sandeepmistry