noble/noble #643: Peripheral connectable detection

samueleforconi, 2017-06-30:
The peripheral.connectable flag is set like this in gap.js:

`  var connectable = (type === 0x04 && previouslyDiscovered) ? this._discoveries[address].connectable : (type !== 0x03);`

I have a beacon that sends two types of frames:
1. a pure iBeacon frame every 100ms (`ADV_NONCONN_IND`)
2. a custom frame with vendor data every 2s (`ADV_IND`), **with scan response**

In my javascript code I filter the advertisement data, so I can detect when the beacon is sending the iBeacon packet and when is sending the custom packet, anyway the `connectable` flag is not toggled to `true` when the beacon is receiving the custom packet.

I think that this is what's happening:
the beacon is firstly recognized when it sends the iBeacon packet (because it sends this packet every 100ms), so the initial `connectable` flag is set to `false`.
When the beacon sends the custom packet, with `ADV_IND` and **scan response**, the `connectable` flag is still set to the previous value because `type === 0x04` (scan response) and `previouslyDiscovered` is `true`.

Could this be a valid assumption of why I do not see the connectable flag toggling?

sandeepmistry, 2017-07-31:
@samueleforconi 

> Could this be a valid assumption of why I do not see the connectable flag toggling?

Yes, looks like noble doesn't take this scenario into account, would you be able to submit a PR to address this?

homedad, 2017-10-04:
@samueleforconi and @sandeepmistry 

I have a BLE device that uses **ADV_SCAN_IND** (not connectable, but with scan response).

I think that the connectable computation in gap.js should be using the flags in the PDU type 01. But those flags aren't being pulled from the eir. The packet type, which is being used, seems to have a different meaning.

The result is that my **ADV_SCAN_IND** device reports being connectable when it is not.

Have either of you made progress on the PR suggested in this thread? Or do you have any other suggestions?