noble/noble #645: N-API support in Node.js 8

vshymanskyy, 2017-07-04:
Are there any plans to support N-API? I think it can **drastically** simplify binary packages deployment ;)

jasongin, 2017-07-08:
Hi, I'm one of the developers on the N-API project. Coincidentally I've recently been working with noble as well (via [noble-uwp](https://github.com/sandeepmistry/noble/issues/550)), with completely unrelated motivations.

While I think it's up to @sandeepmistry to answer the question about plans for this project, I just wanted to say I'm available to answer any questions about N-API. And in case anyone here doesn't know what we're talking about, [here is a blog post introducing N-API](https://medium.com/the-node-js-collection/n-api-next-generation-node-js-apis-for-native-modules-169af5235b06).

Technically, the `noble` package itself doesn't contain any native code, so N-API doesn't apply directly to it, though it could be relevant for the dependency packages (`bluetooth-hci-socket` and `xpc-connection`) that are native code.

sandeepmistry, 2017-07-31:
@vshymanskyy please see @jasongin's comment above.

To get any movement on this, you'll have to open a pull request or issue on node-bluetooth-hci-socket and node-xpc-connection, noble itself doesn't contain any native modules.

closed 2017-07-31 by sandeepmistry