noble/noble #649: peripheral.advertisement.localName undefined with Raspberry Pi 3

davecrumbacher, 2017-07-14:
I am using an [RFduino](http://www.rfduino.com/product/rfd22301-rfduino-ble-smt/index.html) and when using the advertisement-discovery.js I utility get this:

```
peripheral discovered (5408f962e92f4ff18ac480d651b93661 with address <f9:b1:a8:43:84:0f, unknown>, connectable true, RSSI -46:
	hello my local name is:
		RFduino
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"4c000215b9407f30f5f8466eaff925556b57fe6d00010001c5"
```
However, when I do the same thing on a Raspberry Pi 3, I get this:

```
peripheral discovered (f9b1a843840f with address <f9:b1:a8:43:84:0f, random>, connectable true, RSSI -51:
	hello my local name is:
		undefined
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"4c000215b9407f30f5f8466eaff925556b57fe6d00010001c5"
```

The peripheral.advertisement.localName is undefined and, more importantly, I am not able to communicate with the RFduino, even if I match on the manufacturing data. I am able to connect to it via gatttool, but sending or receiving data does not work. However, it works just fine when running my noble-based Node app on a Mac. Any ideas?

davecrumbacher, 2017-07-14:
I also ran peripheral-explorer.js on the address and get different results between my Mac and the Raspberry Pi:

Mac:
```
(mac) $ node node_modules/noble/examples/peripheral-explorer.js f9:b1:a8:43:84:0f
peripheral with ID 5408f962e92f4ff18ac480d651b93661 found
  Local Name        = RFduino
  Manufacturer Data = 4c000215b9407f30f5f8466eaff925556b57fe6d00010001c5
  Service Data      =
  Service UUIDs     =

services and characteristics:
2220
  2221
    properties  read, notify
    value        | ''
  2222
    properties  writeWithoutResponse, write
  2223
    properties  writeWithoutResponse, write
```

Raspberry Pi:
```
(pi) $ node node_modules/noble/examples/peripheral-explorer.js f9:b1:a8:43:84:0f
peripheral with ID f9b1a843840f found
  Manufacturer Data = 4c000215b9407f30f5f8466eaff925556b57fe6d00010001c5
  Service Data      =
  Service UUIDs     =

services and characteristics:
noble warning: unknown peripheral f9b1a843840f
```

davecrumbacher, 2017-07-14:
FYI, I have run `hcidump -t -x` for both commands on the Raspberry Pi (advertisement-discovery.js and peripheral-explorer.js) and have attached the output.

[dump1.txt](https://github.com/sandeepmistry/noble/files/1149178/dump1.txt)
[dump2.txt](https://github.com/sandeepmistry/noble/files/1149177/dump2.txt)

davecrumbacher, 2017-07-18:
I was able to mostly resolve this by applying some patches to the Raspberry Pi --
 https://stackoverflow.com/a/42458091/587786. The challenge was the on-board bluetooth controller on the Raspberry Pi 3.

closed 2017-07-18 by davecrumbacher

nbfuhao, 2017-12-06:
@davecrumbacher I'm facing the same problem where localName is undefined. Can you provide more details about how to fix it?