noble/noble #650: FAIL IN INSTALLATION

NervenCid, 2017-07-15:
Hello to all:

I tried to install noble in windows 10 with Node 8 and i got this:

gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\nerve\node_modules\npm\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:125:13)
gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:197:12)
gyp ERR! System Windows_NT 10.0.15063
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\nerve\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\noble\\node_modules\\usb\\src\\binding\\usb_bindings.node" "--module_name=usb_bindings" "--module_path=C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\noble\\node_modules\\usb\\src\\binding"
gyp ERR! cwd C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb
gyp ERR! node -v v8.1.4
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\nerve\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:125:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:897:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:208:5)
node-pre-gyp ERR! System Windows_NT 10.0.15063
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\noble\\node_modules\\usb\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb
node-pre-gyp ERR! node -v v8.1.4
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! not ok
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\nerve\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding' (1)
npm ERR! path C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\node_modules\node-pre-gyp\node_modules\tar-pack\node_modules\debug\node_modules
npm ERR! code EPERM
npm ERR! errno -4048
npm ERR! syscall scandir
npm ERR! Error: EPERM: operation not permitted, scandir 'C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\node_modules\node-pre-gyp\node_modules\tar-pack\node_modules\debug\node_modules'
npm ERR!  { Error: EPERM: operation not permitted, scandir 'C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\node_modules\node-pre-gyp\node_modules\tar-pack\node_modules\debug\node_modules'
npm ERR!   errno: -4048,
npm ERR!   code: 'EPERM',
npm ERR!   syscall: 'scandir',
npm ERR!   path: 'C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\noble\\node_modules\\usb\\node_modules\\node-pre-gyp\\node_modules\\tar-pack\\node_modules\\debug\\node_modules' }
npm ERR!
npm ERR! Please try running this command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\nerve\AppData\Roaming\npm-cache\_logs\2017-07-15T02_50_03_750Z-debug.log

Thank you

istitou, 2017-07-17:
It seems like you need to run this as an administrator. Search for cmd.exe on Windows, right click, and choose 'Run as Administrator' and that should solve it

NervenCid, 2017-07-17:
Hello and thank you for the answer: 

I run the command as administrator

I run again and i got:

> usb@1.2.0 install C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ 'C:\\Program Files\\nodejs\\node.exe',
node-pre-gyp verb cli   'C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\noble\\node_modules\\usb\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.30
node-pre-gyp info using node@8.1.4 | win32 | x64
node-pre-gyp verb command install []
node-pre-gyp info check checked for "C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding\usb_bindings.node" (not found)
node-pre-gyp http GET https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-win32-x64.tar.gz
node-pre-gyp http 404 https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-win32-x64.tar.gz
node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@8.1.4 (node-v57 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http Pre-built binary not available for your system, looked for https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-win32-x64.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "C:\Program Files\Python36\python.EXE", you can set the PYTHON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (C:\Users\nerve\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\configure.js:483:19)
gyp ERR! stack     at PythonFinder.<anonymous> (C:\Users\nerve\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\configure.js:508:16)
gyp ERR! stack     at C:\Users\nerve\AppData\Roaming\npm\node_modules\npm\node_modules\graceful-fs\polyfills.js:284:29
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:152:21)
gyp ERR! System Windows_NT 10.0.15063
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "--fallback-to-build" "--module=C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\noble\\node_modules\\usb\\src\\binding\\usb_bindings.node" "--module_name=usb_bindings" "--module_path=C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\noble\\node_modules\\usb\\src\\binding"
gyp ERR! cwd C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb
gyp ERR! node -v v8.1.4
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\nerve\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:125:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:897:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:208:5)
node-pre-gyp ERR! System Windows_NT 10.0.15063
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\noble\\node_modules\\usb\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb
node-pre-gyp ERR! node -v v8.1.4
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! not ok
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\nerve\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding' (1)
npm ERR! path C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\node_modules\node-pre-gyp\node_modules\tar-pack\node_modules\fstream-ignore\node_modules\minimatch\node_modules
npm ERR! code EPERM
npm ERR! errno -4048
npm ERR! syscall scandir
npm ERR! Error: EPERM: operation not permitted, scandir 'C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\node_modules\node-pre-gyp\node_modules\tar-pack\node_modules\fstream-ignore\node_modules\minimatch\node_modules'
npm ERR!  { Error: EPERM: operation not permitted, scandir 'C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\node_modules\node-pre-gyp\node_modules\tar-pack\node_modules\fstream-ignore\node_modules\minimatch\node_modules'
[... cut ...]
npm ERR!   code: 'EPERM',
npm ERR!   syscall: 'scandir',
npm ERR!   path: 'C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\noble\\node_modules\\usb\\node_modules\\node-pre-gyp\\node_modules\\tar-pack\\node_modules\\fstream-ignore\\node_modules\\minimatch\\node_modules' }
npm ERR!
npm ERR! Please try running this command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\nerve\AppData\Roaming\npm-cache\_logs\2017-07-17T05_00_29_185Z-debug.log

NervenCid, 2017-07-17:
I installed Python2.7 and intalled under admin but i got:

C:\Windows\system32>npm install -g noble

> usb@1.2.0 install C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ 'C:\\Program Files\\nodejs\\node.exe',
node-pre-gyp verb cli   'C:\\Users\\nerve\\AppData\\Roaming\\npm\\node_modules\\noble\\node_modules\\usb\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.30
node-pre-gyp info using node@8.1.4 | win32 | x64
node-pre-gyp verb command install []
node-pre-gyp info check checked for "C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding\usb_bindings.node" (not found)
node-pre-gyp http GET https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-win32-x64.tar.gz
node-pre-gyp http 404 https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-win32-x64.tar.gz
node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@8.1.4 (node-v57 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http Pre-built binary not available for your system, looked for https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-win32-x64.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
Los proyectos de esta solución se van a compilar de uno en uno. Para habilitar la compilación en paralelo, agregue el modificador "/m".
  core.c
  descriptor.c
  hotplug.c
  io.c
  strerror.c
  sync.c
  poll_windows.c
  threads_windows.c
C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2
011: 'timespec': nueva definición del tipo 'struct' (compilando archivo de origen ..\libusb\libusb\hotplug.c) [C:\Users
\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\build\libusb.vcxproj]
C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2
011: 'timespec': nueva definición del tipo 'struct' (compilando archivo de origen ..\libusb\libusb\descriptor.c)C:\User
s\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 't
imespec': nueva definición del tipo 'struct' (compilando archivo de origen ..\libusb\libusb\core.c)C:\Users\nerve\AppDa
ta\Roaming\npm\node_modules\noble\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': nue
va definición del tipo 'struct' (compilando archivo de origen ..\libusb\libusb\sync.c) [C:\Users\nerve\AppData\Roaming\
npm\node_modules\noble\node_modules\usb\build\libusb.vcxproj]

C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2
011: 'timespec': nueva definición del tipo 'struct' (compilando archivo de origen ..\libusb\libusb\os\poll_windows.c) [
C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\build\libusb.vcxproj]
C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2
011: 'timespec': nueva definición del tipo 'struct' (compilando archivo de origen ..\libusb\libusb\io.c)C:\Program File
s (x86)\Windows Kits\10\Include\10.0.15063.0\ucrt\time.h(39): note: vea la declaración de 'timespec' (compilando archiv
o de origen ..\libusb\libusb\hotplug.c)C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\libusb\li
busb\os/threads_windows.h(47): error C2011: 'timespec': nueva definición del tipo 'struct' (compilando archivo de orige
n ..\libusb\libusb\strerror.c) [C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\build\libusb.vcx
proj]

C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2
011: 'timespec': nueva definición del tipo 'struct' (compilando archivo de origen ..\libusb\libusb\os\threads_windows.c
) [C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.15063.0\ucrt\time.h(39): note: vea la declaración de 'timespec' (
  compilando archivo de origen ..\libusb\libusb\descriptor.c)

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.15063.0\ucrt\time.h(39): note: vea la declaración de 'timespec' (
  compilando archivo de origen ..\libusb\libusb\core.c)
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.15063.0\ucrt\time.h(39): note: vea la declaración de 'timespec' (
  compilando archivo de origen ..\libusb\libusb\io.c)C:\Program Files (x86)\Windows Kits\10\Include\10.0.15063.0\ucrt\t
  ime.h(39): note: vea la declaración de 'timespec' (compilando archivo de origen ..\libusb\libusb\strerror.c)

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.15063.0\ucrt\time.h(39): note: vea la declaración de 'timespec' (
  compilando archivo de origen ..\libusb\libusb\os\poll_windows.c)C:\Program Files (x86)\Windows Kits\10\Include\10.0.1
  5063.0\ucrt\time.h(39): note: vea la declaración de 'timespec' (compilando archivo de origen ..\libusb\libusb\os\thre
  ads_windows.c)

[... 36 lines cut ...]
[... cut ...]
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\nerve\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=C:\Users\nerve\AppData\Roaming\npm\node_modules\noble\node_modules\usb\src\binding' (1)
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.2.0 (node_modules\noble\node_modules\usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.2.0 install: `node-pre-gyp install --fallback-to-build`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

abrowdi, 2017-07-26:
I have the same problem on Win7. What works for me is to use node 6.11.1.
Probably not the solution you are looking for :-(

sandeepmistry, 2017-07-31:
I suggest following here: https://github.com/tessel/node-usb

closed 2017-07-31 by sandeepmistry