noble/noble #654: bluetooth stuff with pi Zero W

neruve, 2017-07-19:
Hello, i am using this with homebridge and switchmates. However it seems the pre-req for this doesn't work with the pi zero w.

Has anyone encountered this and have a fix?

sandeepmistry, 2017-07-31:
@neruve I have it running on my Pi Zero W with no problem. Please re-open and provide more details.

closed 2017-07-31 by sandeepmistry

nikhilmehta15, 2019-08-29:
@sandeepmistry : I am trying to use noble in nodejs code. while executing the code on pi zero w. I am facing below error:

/node_modules/bluetooth-hci-socket/build/Release/binding.node: wrong ELF class: ELFCLASS64
at Object.Module._extensions..node (module.js:681:18)
at Module.load (module.js:565:32)
at tryModuleLoad (module.js:505:12)
at Function.Module._load (module.js:497:3)
at Module.require (module.js:596:17)
at require (internal/module.js:11:18)
at Object. (/home/pi/Development/BleServer/node_modules/bluetooth-hci-socket/lib/native.js:3:15)
at Module._compile (module.js:652:30)
at Object.Module._extensions..js (module.js:663:10)
at Module.load (module.js:565:32)