noble/noble #656: more contributors/maintainers?

jrobeson, 2017-07-20:
@sandeepmistry : You seem pretty busy. Is there any change you might want to bring some folks on board to help out with noble?  I can't say I can contribute a ton myself, but I'd definitely help out with some issue triaging and the like.

Is there anybody here that you feel you can trust to merge code into noble proper though? I personally wouldn't trust myself to merge anything but the most trivial patches, but some other folks here have done good work.

jrobeson, 2017-07-20:
ah, i didn't realize I had the contributor mark.   What do you consider the policy here?

sandeepmistry, 2017-09-05:
@jrobeson currently myself, @jacobrosenthal, and @monteslu have write access to this repo.

> I can't say I can contribute a ton myself, but I'd definitely help out with some issue triaging and the like.

That would be awesome! Let use know if you are still interested in this. Thanks.

jrobeson, 2018-08-12:
I sent you a mail because i forgot i wrote this bug. @sandeepmistry I wrote you an email, did you get it?

rzr, 2019-02-28:
Co-maintainers positions are open:
https://github.com/abandonware/noble/issues/9

Please review pull requests at:
https://github.com/abandonware/noble/pulls

rzr, 2021-01-26:
FYI I just released a downstream release: 

https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-11

Please also comment:

https://github.com/abandonware/noble/issues/2

rzr, 2021-03-21:
@sandeepmistry, Are you interested to relocate this repo to:
https://github.com/adoptware/

For community co-maintenance without scarifying security? see:
https://rzr.github.io/rzr-presentations/docs/abandonware

Then we'll merge @abandonware changes in like proposed at:
https://github.com/abandonware/noble/issues/2

rzr, 2024-02-06:
New release at:

https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-24

Thanks @donavanbecker, feel free to pick his changes (And dependencies)

https://github.com/abandonware/noble/pulls?q=is%3Apr+author%3Adonavanbecker+is%3Aclosed