noble/noble #657: services (and uuid?) of peripheral different between macos and linux

chee, 2017-07-21:
this: `(linux)`
[image]

and this: `(macOS)`
[image]

are the same peripheral, a Bluetooth Low Energy lightbulb. but when querying it from Linux i get 4 services, while i only see 3 on the macOS side.
they also both have a different value for uuid? perhaps this is normal, i don't have any other BLE devices around here to talk to, it's just me and the blub.
both are using the same version of noble, the Linux side is `arch linux arm` on a Raspberry Pi 3.

it's the same small script on both machines, which you can find in the comment below.

chee, 2017-07-21:
```js
const noble = require('noble')

noble.on('stateChange', state =>
  state === 'poweredOn' && discover())

const discover = () => {
  noble.on('discover', peripheral => {
    if (peripheral.address !== '00:21:4d:03:20:1b') return
    noble.stopScanning()
    console.log(peripheral.uuid)
    peripheral.connect(() => interrogate(peripheral))
  })

  noble.startScanning()
}

const interrogate = peripheral =>
  peripheral.discoverAllServicesAndCharacteristics(() => console.log(list(peripheral)))

const list = peripheral =>
 peripheral.services.map((service, index) =>
   `${index}. ${service.uuid}`
 ).join('\n')
```

sandeepmistry, 2017-07-31:
@chee macOS blocks the `1800` service. It's better to compare the UUID's.

closed 2017-07-31 by sandeepmistry

chee, 2017-08-07:
thank you a lot, that has worked fine. i now can control my bluetooth bulbs with my javascript computer.