noble/noble #658: Error: Command Disallowed when connecting on Raspberry Pi

lrucker1, 2017-07-22:
I got this code working on MacOS : https://github.com/lrucker1/homebridge-dotti
but when I ran it on Raspberry Pi it had several problems:
first, "startScanning([DottiServiceUuid], false)" never found the Dotti device.
I tweaked it to accept everything and filtered it by device address instead, then it hit "Command Disallowed"

Raspberry Pi 3, "Raspbian GNU/Linux 8 (jessie)" , everything freshly installed.

"sudo homebridge -D"
[7/22/2017, 12:00:40 AM] Homebridge is running on port 51828.
[7/22/2017, 12:00:40 AM] [Dotti] Starting Noble scan..
[7/22/2017, 12:00:40 AM] [Dotti] Found accessory cbcb2f1752a1, address cb:cb:2f:17:52:a1, expected 5c:31:3e:8b:26:a4. Skipping...
[7/22/2017, 12:00:40 AM] [Dotti] Found accessory e10afefebb69, address e1:0a:fe:fe:bb:69, expected 5c:31:3e:8b:26:a4. Skipping...
[7/22/2017, 12:00:40 AM] [Dotti] Found accessory 5c313e8b26a4, address 5c:31:3e:8b:26:a4, connecting..
[7/22/2017, 12:00:40 AM] [Dotti] Noble connection failed: Error: Command Disallowed (0xc)

Full debug output and hcidump attached:

[hcidumpOutput.txt](https://github.com/sandeepmistry/noble/files/1167171/hcidumpOutput.txt)
[hciDump.txt](https://github.com/sandeepmistry/noble/files/1167176/hciDump.txt)
[homebridgeRun.txt](https://github.com/sandeepmistry/noble/files/1167177/homebridgeRun.txt)

lrucker1, 2017-07-22:
Although I had rebooted at least twice and still hit that, the reboot after I filed this worked. Though the code that's uploaded to github still fails because startScanning only works with an empty uuid list.

sandeepmistry, 2017-07-31:
@lrucker1 closing this as resolved. Some Bluetooth adapters don't allow scanning and connecting at the same time. https://github.com/sandeepmistry/noble#adapter-specific-known-issues

closed 2017-07-31 by sandeepmistry