noble/noble #659: Noble was installed on RPi0 but it can't run it

aymanhendawy, 2017-07-26:
I'm working on Raspberry Pi0, try to run noble based application, when I run this cmd
#npm list -g noble , I get 
/opt/nodejs/lib
└── noble@1.8.1 

which mean that noble is already installed, but when I run my application I get this error, why?

module.js:472
    throw err;
    ^

Error: Cannot find module 'noble'
    at Function.Module._resolveFilename (module.js:470:15)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/azure/iot-hub-node-raspberrypi-client-app/ble_azure.js:3:13)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)

aymanhendawy, 2017-07-26:
for more details, I'm following this guide of connecting RPI with Azure,
https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-raspberry-pi-kit-node-get-started

I cloned this code,
git clone https://github.com/Azure-Samples/iot-hub-node-raspberrypi-client-app and updadted with my code which require noble.

sandeepmistry, 2017-07-31:
I suggest installing noble without the `-g` option.

closed 2017-07-31 by sandeepmistry

aymanhendawy, 2017-08-01:
@sandeepmistry, thank you, this what I did, and it solved my problem.