noble/noble #663: Can' t subscribe to characteristics

aplyumpv, 2017-08-13:
Hello,

It seems that subscription to characteristics does not work as "Characteristic found..." is never displayed.
Could anyone please check this?

```javascript
var noble = require('noble');

noble.on('stateChange', function (state) {
  if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', (peripheral) => {
  if (peripheral.id.toUpperCase() == 'B9G8K4S9D9G0') {
    console.log('Connected...');
    peripheral.connect((err) => {
      peripheral.discoverServices(['219739asd78fdsa79f7sasd79afs98f7'], (err, services) => {
        console.log("Service discovered...");
        services.forEach((service) => {
          service.discoverCharacteristics(['78797809808as0df808da0s8df0s8a08'], (err, characteristics) => {
            console.log('Characteristic found...');
            var characteristic = characteristics[0];
            characteristic.on('data', (data) => {
              characteristic.read((err, readData) => {
                console.log(readData.toString('utf-8'));
              });
            });
            characteristic.subscribe(()=> {
				console.log('Subscribed to characteristic ...');
			});
          });
        });
      });
    });
  }
```

 hcidump -t -X
```
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2017-08-13 12:13:54.646212 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2017-08-13 12:13:54.655192 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2017-08-13 12:13:54.670191 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
    LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
    Manufacturer: Cambridge Silicon Radio (10)
2017-08-13 12:13:54.672335 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    0000: 00                                                .
2017-08-13 12:13:54.674171 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    0000: 00 01 00                                          ...
2017-08-13 12:13:54.675182 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:1A:7D:DA:71:13
2017-08-13 12:13:54.740193 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2017-08-13 12:13:54.746202 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2017-08-13 12:13:54.782197 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2017-08-13 12:13:54.784190 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2017-08-13 12:13:54.787169 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2017-08-13 12:13:55.012189 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2017-08-13 12:13:55.039189 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 0
2017-08-13 12:13:55.131169 > HCI Event: Command Status (0x0f) plen 4
    Unknown (0x00|0x0000) status 0x00 ncmd 1
2017-08-13 12:13:55.146160 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```

sandeepmistry, 2017-09-04:
@aplyumpv 

```
if (peripheral.id.toUpperCase() == 'B9G8K4S9D9G0') {
```

This is not likely a valid peripheral id, ad `K` an `G` are not valid characters for a BT address and noble on Linux uses the BT address as the peripheral id. So I would not expect your sample snippet to work at all.

Please try again and provide more info and re-open this issue.

closed 2017-09-04 by sandeepmistry

aplyumpv, 2017-09-11:
Yes, sorry, that was just an example and here is the exact code that is executed. 
It gets to "Characteristic found..." and nothing else happens.

```javascript
var noble = require('noble');

noble.on('stateChange', function (state) {
  if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', (peripheral) => {
  if (peripheral.id.toUpperCase() == 'B2F669A94E45') {
    console.log('Connected...');
    peripheral.connect((err) => {
      peripheral.discoverServices(['49535343fe7d4ae58fa99fafd205e455'], (err, services) => {
        console.log("Service discovered...");
        services.forEach((service) => {
          service.discoverCharacteristics(['495353431e4d4bd9ba6123c647249616'], (err, characteristics) => {
            console.log('Characteristic found...');
            var characteristic = characteristics[0];
            characteristic.on('data', (data) => {
              characteristic.read((err, readData) => {
                console.log(readData.toString('utf-8'));
              });
            });
            characteristic.subscribe(()=> {
				console.log('Subscribed to characteristic ...');
			});
          });
        });
      });
    });
  }
});
```

ronangaillard, 2017-11-15:
Same, I receive notifications on iOs using the NRF Connect app, but using nobble with `subscribe` and `char.on('data', ...)` does not trigger any callback after notification. This is very frustrating.

hkayann, 2021-07-21:
This issue persists. I can subscribe with NRF connect but can not with `characteristic.subscribe()`.