noble/noble #664: Regular disconnect

gotleg, 2017-08-14:
Hello,

I work for many month on a node/electron application to control Parrot Mambo mini-drone via BLE using npm-parrot-minidrone lib on top of noble.

Everything is working really great except that I got a two strange beahaviors when the drone is disconnected/powered off.
Note that this behavior has been reproduced on both my dev env. (Ubuntu 16.04 in Virtual Box) and my production env. (Raspberry pi 3). I'm using Sena UD-100 Bluetooth dongle that allows me to achieve very good range.

1/ using npm-parrot-minidrone
With this implementation there is no disconnect handler on the peripheral and when the drone is disconnected, noble sends me a poweredOff event (not for the drone but for the bluetooth dongle). Then noble is not able to do anything else as it doesn't detect the dongle anymore. Note that my host windows also detects a USB unplug/replug. I also reproduced that PoweredOff event on my RPI3.

Here is the interesting part of the code : 
`
onPeripheralDiscovery(peripheral) {
        if (!this.validatePeripheral(peripheral)) {
            return;
        }
        Logger.info('Peripheral found ${peripheral.advertisement.localName}');
        this.noble.stopScanning();
        peripheral.connect((error) => {
            if (error) {
                throw error;
            }
            this.peripheral = peripheral;
            this.setupPeripheral();
        });
    }
`
 

_All source code and examples are here : [https://github.com/fetherston/npm-parrot-minidrone/](https://github.com/fetherston/npm-parrot-minidrone/)_

2/ As I wanted to have a clean test code (outside of electron and npm-parrot-minirone) I decided to test a simple noble connection to my drone.

This time i'm able to detects when peripheral is disconnected, startScanning, re-discover my peripheral and reconnect. This would be perfect except that once my peripheral has be disconnected the first time, when it reconnects it loose connection after a few seconds and then reconnects, ...
On this test it happens only one time but after many tests I feel like if I add listeners characteristics change the connection is more and more unstable.

Here is the code : 
`var noble = require('noble');
var peripheralIdOrAddress = 'e0:14:fb:7b:3d:d1' //process.argv[0].toLowerCase();
var peripheral = null
var loaded = false
noble.on('stateChange', function(state) {
  console.log('## stateChange : ' + state + ' ##')
  if (state === 'poweredOn') {
    console.log("start scanning")
    noble.startScanning();
  } else {
    console.log("stop scanning")
    noble.stopScanning();
  }
});
noble.on('warning', function (warning) {
  console.log('warning : ' + warning)
})
noble.on('stateChange', function (data) { log('stateChange', data)});
noble.on('addressChange', function (data) { log('addressChange', data)});
noble.on('scanStart', function (data) { log('scanStart', data)});
noble.on('scanStop', function (data) { log('scanStop', data)});
noble.on('discover', function (data) { log('discover', data)});
noble.on('connect', function (data) { log('connect', data)});
noble.on('disconnect', function (data) { log('disconnect', data)});
noble.on('rssiUpdate', function (data) { log('rssiUpdate', data)});
noble.on('servicesDiscover', function (data) { log('servicesDiscover', data)});
noble.on('includedServicesDiscover', function (data) { log('includedServicesDiscover', data)});
noble.on('characteristicsDiscover', function (data) { log('characteristicsDiscover', data)});
noble.on('read', function (data) { log('read', data)});
noble.on('write', function (data) { log('write', data)});
noble.on('broadcast', function (data) { log('broadcast', data)});
noble.on('notify', function (data) { log('notify', data)});
noble.on('descriptorsDiscover', function (data) { log('descriptorsDiscover', data)});
noble.on('valueRead', function (data) { log('valueRead', data)});
noble.on('valueWrite', function (data) { log('valueWrite', data)});
noble.on('handleRead', function (data) { log('handleRead', data)});
noble.on('handleWrite', function (data) { log('handleWrite', data)});
noble.on('handleNotify', function (data) { log('handleNotify', data)});
function log(message, data) {
  console.log(message + ' : ' + data)
}
noble.on('discover', function(peripheral) {
  console.log('## discover ' + peripheral + ' ##')
  //if (peripheral.id === peripheralIdOrAddress || peripheral.address === peripheralIdOrAddress) {
    noble.stopScanning();

    console.log('peripheral with ID ' + peripheral.id + ' found');

    var advertisement = peripheral.advertisement;
    var localName = advertisement.localName;
    var txPowerLevel = advertisement.txPowerLevel;
    var manufacturerData = advertisement.manufacturerData;
    var serviceData = advertisement.serviceData;
    var serviceUuids = advertisement.serviceUuids;

    if (localName) {
      console.log('  Local Name        = ' + localName);
    }
    if (txPowerLevel) {
      console.log('  TX Power Level    = ' + txPowerLevel);
    }
    if (manufacturerData) {
      console.log('  Manufacturer Data = ' + manufacturerData.toString('hex'));
    }
    if (serviceData) {
      console.log('  Service Data      = ' + serviceData);
    }
    if (serviceUuids) {
      console.log('  Service UUIDs     = ' + serviceUuids);
    }
    init(peripheral);
  //}
});
function init(peripheral) {
  console.log('## init ##')
  if (!loaded) {
    loaded = true
    peripheral.on('disconnect', function() {
      console.log("disconnected")
      noble.startScanning();
    });
    peripheral.on("connect", function() {
      console.log("Connected")
    })
  }
  peripheral.connect(function(error) {
    if (error) {
      console.log(error)
    }
  })
}`

As you can see I tried to get the maximum of info from noble but in the app log I just get : Connected -> Disconnected.

Any idea of why this disconnect event is raised ?

How to get more log to go further and undestand this behavior ?

We made some tests today on MacOS and the behavior was not exactly the same ?

gotleg, 2017-08-15:
To complete this issue here is an HCI dump : 
```
HCI sniffer - Bluetooth packet analyzer ver 5.41
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
2017-08-14 19:50:51.525783 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2017-08-14 19:50:51.528752 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2017-08-14 19:50:51.543343 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x2031
    LMP Version: 4.0 (0x6) LMP Subversion: 0x2031
    Manufacturer: Cambridge Silicon Radio (10)
2017-08-14 19:50:51.546836 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    0000: 00                                                .
2017-08-14 19:50:51.548700 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    0000: 00 01 00                                          ...
2017-08-14 19:50:51.550759 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:01:95:39:8E:C4
2017-08-14 19:50:52.284324 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2017-08-14 19:50:52.287935 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2017-08-14 19:50:52.444451 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2017-08-14 19:50:52.446727 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2017-08-14 19:50:52.449751 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2017-08-14 19:50:52.608802 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2017-08-14 19:50:52.615208 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2017-08-14 19:50:52.773885 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
2017-08-14 19:50:54.883754 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2017-08-14 19:50:56.750568 > HCI Event: Command Complete (0x0e) plen 7
    Read RSSI (0x05|0x0005) ncmd 1
    status 0x00 handle 76 rssi -28
2017-08-14 19:50:59.754903 > HCI Event: Command Complete (0x0e) plen 7
    Read RSSI (0x05|0x0005) ncmd 1
    status 0x00 handle 76 rssi -28
2017-08-14 19:51:00.467002 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 76 reason 0x13
    Reason: Remote User Terminated Connection
2017-08-14 19:51:00.522911 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2017-08-14 19:51:00.524903 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2017-08-14 19:51:00.529095 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2017-08-14 19:51:01.177953 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2017-08-14 19:51:01.182949 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2017-08-14 19:51:01.342591 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
2017-08-14 19:51:02.740052 > HCI Event: Command Complete (0x0e) plen 7
    Read RSSI (0x05|0x0005) ncmd 1
    status 0x00 handle 74 rssi -28
2017-08-14 19:51:03.342097 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2017-08-14 19:51:04.331044 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2017-08-14 19:51:04.335179 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2017-08-14 19:51:04.343154 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 74 reason 0x16
    Reason: Connection Terminated by Local Host
2017-08-14 19:51:04.390841 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2017-08-14 19:51:04.393959 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2017-08-14 19:51:04.397127 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```

sandeepmistry, 2017-09-05:
@gotleg have you had a chance to try another Bluetooth 4.0 adapter?

gotleg, 2017-10-14:
@sandeepmistry Yes and my problems where the same.

I'll come back to my previous post. I found a solution and I wanted to share.

My initial problem was in npm-parrot-minidrone implementation there was no disconnect handler. As soon as I tried to handle disconnect, the Bluetooth adapter went to state "poweredOff" each time my peripheral got disconnected.
I finally discovered that we write some characteristics every 100-300ms (for drone flight params) and when we try to access to characteristics of the peripheral that is beeing disconnected, Boom dongle poweredOff. 
I can "solve" the problem by increasing the write loop to 500ms. This is a real problem for us as we are using bluetooth to pilot a drone :/
But I think these 500ms allows noble to handle disconnect and avoid the problem.