noble/noble #665: TypeError: listener must be a function on Raspberry Pi

botaksetan, 2017-08-22:
Hi,

I have created a file called noblescan.js as below:

```
var noble = require('noble');

noble.on('discover', function(peripheral) {

var macAddress = peripheral.uuid;
var rss = peripheral.rssi;
var localName = advertisement.localName;
console.log('found device: ', macAdress, ' ', localName, ' ', rss);
})

noble.startScanning.bind();
```

But, when it was executed with command node noblescan.js it was throwing error as below:
```

events.js:197
    throw new TypeError('listener must be a function');
    ^

TypeError: listener must be a function
    at Noble.addListener (events.js:197:11)
    at Object.<anonymous> (/home/pi/noblescan.js:3:7)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Function.Module.runMain (module.js:467:10)
    at startup (node.js:134:18)
    at node.js:961:3
```

Please somebody help me with this problem.

Thanks,

Adi

cruepprich, 2017-08-28:
I've seen the same thing on my Pi 3. I had it working on an older Pi a year ago and I think it may be incompatible with the newer version of Jessie.

sandeepmistry, 2017-09-05:
@botaksetan I think the `noble.startScanning.bind();` is the root cause, please try the advertisement discovery example: https://github.com/sandeepmistry/noble/blob/master/examples/advertisement-discovery.js

and let us know how it goes.

closed 2017-09-05 by sandeepmistry

botaksetan, 2017-09-05:
@sandeepmistry yes, it was `noble.startScanning.bind();` that caused the problem. Thanks to point it out :)

Regards,

Adi